libretro/Lakka-LibreELEC #141: Hotkey to control screen brightness not working - Lenovo Thinkpad w520

CarlUpright, 2017-06-13:
I'm using Build Apr 24 2017 on Linux (Lakka live usb) on a Lenovo Thinkpad w520.
When i press volume - and + , or when i press Fn + Home or Fn + End (supposed to be the brightness control hotkeys), nothing happen. 

I made a hotkey for the volume control, so that's no big deal, but my screen is dimmed to the minimum.

Is there a little script i could use to create a hotkey to control the screen brightness? or maybe i need to use a different display driver ?

Kivutar, 2017-06-13:
It's certainly the kind of screen where a systemd unit is setting the brightness during the boot.

https://www.google.fr/search?q=Lenovo+Thinkpad+w520+linux+screen+brightness&ie=utf-8&oe=utf-8&client=firefox-b&gfe_rd=cr&ei=l5o_WZOgLKLVXr6igfAP

You can try to add `acpi_osi="!Windows 2012" thinkpad-acpi.brightness_enable=1` in the syslinux configuration files on the first partiton of the USB.

If it's not enough, try this command in the CLI, maybe using SSH from another PC:

    echo 100 > /proc/acpi/video/VID1/LCD0/brightness

CarlUpright, 2017-06-13:
i found out that this command works in the CLI using SSH : 
`echo XXX > /sys/class/backlight/intel_backlight/brightness`

Where XXX is a value between 0 (backlight off) and max_brightness (4437 for this particular model).

So it seams like it is really the Fn hotkey that isn't working with Lakka (no problem in win 7, or Ubuntu). Is there a way to create a hotkey in Lakka to send the command to max out the display ? or to create a new value in the menu ?

Kivutar, 2017-06-13:
For now let try the autostart service to automatically set the brightness on boot.

CarlUpright, 2017-06-13:
i don't know how to do that, can you assist me ?

Kivutar, 2017-06-13:
Yes, first read this http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/

CarlUpright, 2017-06-13:
that's already done and i'm there :)

Kivutar, 2017-06-13:
OK, use nano to create the autostart.sh

    nano .config/autostart.sh

Paste the content:

    echo 4437 > /sys/class/backlight/intel_backlight/brightness

And save with CTRL+X

Then enable the service:

    systemctl enable autostart

And reboot to check if it worked

CarlUpright, 2017-06-13:
Done. It is now booting at maximum brightness. That's good enough for me on that laptop, I'll try to boot it on different PC to see if that still works. 

On the side, i think i need to compile the ThinkPad ACPI Extras Driver to activate all my Fn keys. I never compiled anything.

Thank you !

Kivutar, 2017-06-13:
Yeah forget it, in Lakka compiling doesn't work exactly like on the other linux distributions. If this workaround works for you, it's already lucky :)

closed 2017-06-13 by Kivutar

ricsif, 2021-05-23:
@kivutar @CarlUpright 
Thanks for the steps, it solved similar brightness problem on Lenovo T430 (I have two, faced the issue only on one of them).
Despite the following error, after reboot .config/autostart.sh was executed.
```
Lakka:~ # systemctl enable autostart
Failed to enable unit: File autostart.service: No such file or directory
```

hailingfang, 2024-11-16:
> @kivutar @CarlUpright Thanks for the steps, it solved similar brightness problem on Lenovo T430 (I have two, faced the issue only on one of them). Despite the following error, after reboot [...]
```
echo '#!/bin/bash' > /storage/.config/autostart.sh
echo 'echo 100 > /sys/class/backlight/intel_backlight/brightness' >> /storage/.config/autostart.sh
chmod +x /storage/.config/autostart.sh

```

Above codes work on my computer.