libretro/Lakka-LibreELEC #142: Request: include Sixad for Odroid C1

0lace, 2017-06-15:
- Which version of Lakka are you using?
Lakka-OdroidC1.arm-2.1-rc2.
- What system hardware are you using?
Odroid C1.
- What did you do?
Try to pair DS3 controller with sixad.
- What did you expect to happen?
Pair correct DS3 controller.
- What happened instead?
Nothing, command not found. 

`Lakka:~ # sixpair
Current Bluetooth master: 68:14:01:a6:4c:7c
Setting master bd_addr to 00:18:e4:0a:2e:6b
Lakka:~ # sixad --start
-sh: sixad: not found `

The steps I want to do is described here:
http://www.pabr.org/sixlinux/sixlinux.en.html
https://help.ubuntu.com/community/Sixaxis
https://github.com/RetroPie/RetroPie-Setup/wiki/PS3-Controller

Its important to make one link to one file were we can edit to add the following configurations:

#/etc/bluetooth/hcid.conf 
#or what us choose better

`iscan disable;
pscan enable;
#auth enable;                                        
#encrypt enable; `

Kivutar, 2017-06-22:
Sixad adds a lot of things in the system: https://github.com/RetroPie/sixad/blob/master/Makefile

We need to figure out the parts we really need, and create a systemd unit for it.

Also, you have to know that once you use sixad, the default bluez path will stop working. It's possible that you won't be able to use a non-ps3 bluetooth controller alongside your DS3.

0lace, 2017-06-25:
The way how they manage sixad is great and, for C1 this is only way to get the second controller to work.

'the default bluez path will stop working.' but is for a great cause, this is no problem for me, I only use bluetooth to play with DS3 controllers.

"t's possible that you won't be able to use a non-ps3" : no problem ;)

closed 2020-08-18 by Ntemis