libretro/Lakka-LibreELEC #143: Can't mount /dev/mmcblk after reboot Odroid C1.

0lace, 2017-06-15:
- Which version of Lakka are you using?
Lakka-OdroidC1.arm-2.1-rc2.
- What system hardware are you using?
Odroid C1.
- What did you do?
reboot from GUI or CLI.
- What did you expect to happen?
To load Lakka.
- What happened instead?
Stop after reboot with one message: "Can't mount /dev/mmcblk "
[image]

Kivutar, 2017-06-15:
Usually this is due to a flashing corruption. Try reflashing 2 or 3 times using Etcher.

0lace, 2017-06-15:
@Kivutar I have used dd on the first time.
I have tried Etcher, 3 times and the result is the same.

Kivutar, 2017-06-16:
You're using an SD or an eMMC?

0lace, 2017-06-16:
Hi, I'm using SD.

Kivutar, 2017-06-16:
I could reproduce the problem. RC2 and RC3 are indeed broken for C1. Was it also the case with 2.1-rc1?

0lace, 2017-06-16:
I have tested the RC2 only, I could test and post the result here.

Kivutar, 2017-06-16:
Yes please, It would save me time :)

Kivutar, 2017-06-18:
It doesn't happen with RC1

Kivutar, 2017-06-19:
Ah, I flashed on another SD card, and this time 2.1-rc2 worked!!!

closed 2017-06-24 by Kivutar