libretro/Lakka-LibreELEC #144: Add support for builtin wifi with ap6210 module on BananaPro image

agent94ger, 2017-06-15:
Can you add support for the on-board wifi on Banana Pro boards?
The standard way to activate wifi on other Banana Pro distros would be:
`# modprobe ap6210`
`modprobe: FATAL: Module ap6210 not found in directory /lib/modules/3.4.91`
Seems like it would be as "easy" as adding the module to the image, but I couldn't do it.

Kivutar, 2017-06-15:
Are you talking about the build we're calling Bananapi or H3?

agent94ger, 2017-06-15:
I'm using the Lakka-Bananapi.arm-2.0.img image.

Kivutar, 2017-06-15:
Sadly, adding back ap6210 will not help. Wifi is broken anyway because of an incompatibility between the old 3.4 sunxi kernel and modern systemd versions.

agent94ger, 2017-06-16:
So no easy fix? That's really kinda sad...

Kivutar, 2017-06-16:
No real fix. I will still try to add back the ap6210 for you to test. But you will get an error "no carrier" when starting connmanctl.

agent94ger, 2017-06-18:
Thank you for the testing opportunity, just tell me what I need to do

Kivutar, 2017-06-22:
Ok this image has the driver. You need to modprobe it manually. http://static.kivutar.me/Lakka-Bananapi.arm-2.1-rc3.img.gz

agent94ger, 2017-06-23:
Thanks for the image! We got a new error though:
`Lakka:~ # modprobe ap6210`
`modprobe: ERROR: could not insert 'ap6210': No such device`
That errormessage appears on the first run after booting. If I try again, this message appears:
`Lakka:~ # modprobe ap6210`
`modprobe: ERROR: could not insert 'ap6210': Operation not permitted`
What will we try next?

Kivutar, 2017-06-23:
Can you also try this?

    modprobe bcmdhd

agent94ger, 2017-06-23:
That gives me the same message:
`Lakka:~ # modprobe bcmdhd`
`modprobe: ERROR: could not insert 'bcmdhd': No such device`

Kivutar, 2017-06-23:
OK I found something that can help.

According to this link https://forum.armbian.com/index.php?/topic/219-bananapi-pro-debian-wheezy-kernel-34108-wifi-problem/ we need to add something in the fex.

But we need to make sure that I have to apply the ap6210 patch, because there is already bcmdhd in the kernel and it's supposed to be the same driver.

Kivutar, 2017-06-23:
Please try this new image http://static.kivutar.me/Lakka-Bananapi.arm-2.1-rc3.img.gz

I changed the kernel config and I updated the script.bin

agent94ger, 2017-06-24:
Modprobing the bcmdhd driver still gives me the same error...
`Lakka:~ # modprobe bcmdhd`
`modprobe: ERROR: could not insert 'bcmdhd': No such device`

Trying to recreate the "No such device"/"Operation not permitted" error combination gave me an new phenomenon:
`Lakka:~ # modprobe ap6210`
`modprobe: ERROR: could not insert 'ap6210': No such device`
`Lakka:~ # modprobe ap6210`
`Segmentation fault`
A third try seemed successful at first:
`Lakka:~ # modprobe ap6210`
`Lakka:~ # lsmod`
`Module                  Size  Used by`
`ap6210                623323  1`
But I still couldn't scan for/find any wifi networks

closed 2020-08-18 by Ntemis