libretro/Lakka-LibreELEC #145: Stuck on flower logo screen

driagonx, 2017-06-15:
I just downloaded and tried to install the LAKKA on my PC and it worked on one but I tried it on another and it just stuck on the logo screen at boot. The installer seemed to work fine.
Alienware M11x R2
•1.20GHz Intel Core i7 640UM (4MB Cache, max turbo frequency 2.266GHz)
•4GB PC3-8500 DDR3 (2 x 2GB)
•11.6″ WXGA WideHD LED Backlit display (1366×768)
•500GB Samsung EVO SSD Hard Drive
•Nvidia GeForce GT 335M Discrete Graphics (1GB GDDR3)

gouchi, 2017-06-17:
As mentioned in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.0/ISSUE_TEMPLATE.md), please provide some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

gouchi, 2017-08-06:
Do you reproduce your issue with [Lakka 2.1 RC4](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz) ?

closed 2017-10-20 by Kivutar