libretro/Lakka-LibreELEC #150: Dualshock 3 Sixaxis menu scrolling out of control

hooby3dfx, 2017-06-19:
- Which version of Lakka are you using?
2.0
- What system hardware are you using?
Orange Pi
- What did you do?
Pair with Dualshock 3 Sixaxis controller (legit/official)
- What did you expect to happen?
Normal menu navigation behavior
- What happened instead?
The menu scrolls horizontally out of control.

This does not seem to happen with other PS3 knockoff controllers.
Could this just be due to deadzone? If so perhaps the defaults should be higher for this controller.
Or could the sixaxis input be affecting anything?

Kivutar, 2017-06-19:
I also noticed this bug on other hardware. I will try to find which ones and do some comparisons.

cottonstar, 2017-06-24:
I have experienced the same issue with an official PS4 controller on a PC build using 2.0. Also unpredictable Page Down-type behaviour within menus and playlists.

Kivutar, 2017-06-24:
We need to know if it's a system bug or a retroarch bug.

TidderM, 2017-07-04:
I have noticed the same with Odroid XU4. When using sixaxis with usb cable, worked as expected until it was paired with Bluetooth. Then wired and wireless connection with sixaxis controller causes scroll spasm. When wired, the spasm stops if I push the center PlayStation button on the controller, and then I can continue using wired without issue until USB is unplugged from it again.

hydraheim, 2017-07-04:
This is exactly what happens to clone DS3 controllers. I have one and it acts like this.

TidderM, 2017-07-07:
I'm having this issue with official PS3 controller as well as my shanwan controller.

cruzdanilo, 2017-07-22:
Same problem here, RPi3 + official DS3.

erkkijakobson, 2017-08-06:
Lakka 2.0 / 2.1 RC4 on RPi3 with a DS4 - autoscroll triggers only while moving over a few large playlists:
* with none or only small playlists there is no horizontal autoscroll mixed with skippy vertical movement
* if there are many large playlists then a longer menu lag is also introduced to autoscrolling over them
* always reproducible with a wireless DS4, but less often also happens with a wired controller

Moving D-pad a lot or turning off the controller stops the autoscrolling on a random first level menu entry
Workaround is to temporarily disable all (change dir) or only large playlists (move lpl or rename extension)

Sniper435, 2018-02-24:
I'm seeing this same problem on an xu4 using the latest lakka release, 5 different PS3 controllers (all official)

gouchi, 2018-02-25:
May you try to do as explained [here](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6) for DS3.

Thank you.

ToKe79, 2018-07-02:
no response, closing stale issue.

closed 2018-07-02 by ToKe79