libretro/Lakka-LibreELEC #151: Changing audio device doesn't work properly

crayonized, 2017-06-20:
First setting i check after setting up my gamepad is the audio device setting. I switch between the devices using d-pad left/right. Eventually Lakka locks onto one of them and after that I can't switch audio devices anymore.

I'll provide a log if I figure out how, but I don't have multiple PCs anymore (so no SSH).

gouchi, 2017-06-24:
Did you succeed to get some [Lakka system and audio log](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

Thank you.

crayonized, 2017-07-01:
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/1116790/log.txt)
[log2.txt](https://github.com/libretro/Lakka-LibreELEC/files/1116792/log2.txt)
[log3.txt](https://github.com/libretro/Lakka-LibreELEC/files/1116791/log3.txt)

gouchi, 2017-07-01:
Thank you for the log but you forgot audio log. Is to possible to get the output of both commands 

```
aplay -L
grep audio_device /storage/.config/retroarch/retroarch.cfg
```

We know it is not user friendly maybe in the future we will change that for the log. At least, creating a script that will upload automatically the result to pastebin will be easier.

crayonized, 2017-07-01:
aplay -L displays a huge list of devices on screen that is impossible to copy by hand. grep audio_device gave me this:

audio_device = "surround21:CARD=PCH,DEV=0"

It seems to just show the device that got stuck this time. This is not the only device that has gotten stuck though. If I would reflash the USB and try again, another device may get stuck instead. It's pretty random.

Saving a file to USB would be better I think. Automatically uploading to pastebin would be useful for some, but not everyone wants to connect every device to the internet. I normally just connect my desktop PC while everything else is permanently offline. Thanks.

gouchi, 2017-07-01:
If you press Start instead of right and left key it should reset your audio device configuration.

Can you try if you reproduce the issue with [Lakka 2.1 RC3](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc3.img.gz) ?

Thank you.

crayonized, 2017-07-01:
I tried 2.1 RC3 now and the locking issue is still there. Resetting the audio device setting with start button seems to work though.

gouchi, 2017-07-30:
Please test it with next release as we had a [fix](https://github.com/libretro/Lakka-LibreELEC/commit/6c47409e7a249e2a92776175a1f36797e4bdb665) for it.

Thank you.

crayonized, 2017-08-21:
Finally got some time to try it and it seems to be fixed as of RC4.

Kivutar, 2017-08-21:
Good

closed 2017-08-21 by Kivutar