libretro/Lakka-LibreELEC #153: Lakka 2.1 RC2 freezes after a minute or two of usage on RPi2

parklez, 2017-06-21:
- Which version of Lakka are you using?
Lakka 2.1 RC2
- What system hardware are you using?
Raspberry Pi 2, overclocked
- What did you do?
Manual update from 2.1 RC1 to RC2, Booted up the system, froze after two minutes, no inputs whatsoever. sometimes the system hangs up on flower as well. it will also freeze while playing anything.
- What did you expect to happen?
No freezes whatsoever, just like previous version (2.1 RC1)
- What happened instead?
Static frame of the moment the system froze.

Kivutar, 2017-06-22:
@Brunnis @larskj do you think it can be due to the recent changes on the governor + RT?

larskj, 2017-06-22:
I doubt it, it might however be caused by the performance governor in combination with an unstable overclock. 

I would suggest testing it again at stock settings, no OC.

parklez, 2017-06-22:
Fresh install of Lakka 2.1 RC2, stock settings, freezes after 2 minutes, no activity whatsoever (no network/no controller inputs). is there a way I could provide detailed info? thanks.

Kivutar, 2017-06-23:
I just tried on my RPi3. I did get a freeze, but I had to wait 1h30.

larskj, 2017-06-23:
If the builds are available and it can fail within minutes, then the easiest is probably just to test the most major commits between the RC1 and RC2 and figure out when the freeze was introduced and then go from there?

Kivutar, 2017-06-23:
From what I see in our changelog http://www.lakka.tv/articles/2017/06/20/lakka-21-rc2-published/
It is either:

 - the kernel upgrade to 4.9.29 
 - the RT patch
 - the governor

I suspect the governor. Sadly, waiting 1h30 for a freeze is not what I would call "within minutes".

@parklez can you confirm that RC1 has no freeze issues at all?

larskj, 2017-06-23:
I agree, waiting 1h30 for a failure is not feasible, but if you can link to the three builds mentioned we can hope @parklez will test them for us on his RPi2 which hopefully fails fast?

parklez, 2017-06-23:
2.1 RC1 was going strong, but I'll do it. I'll take some notes from both fresh installs of "Lakka-RPi2.arm-2.1-rc1.img.gz" and "Lakka-RPi2.arm-2.1-rc2.img.gz" at stock. if they're stable alone, I'll play snes or something then overclock to the previous settings. I'll also do some rebooting just to make sure.

parklez, 2017-06-23:
2.1 RC1 on a fresh install at stock settings works flawlessly. few reboots, played for a while, perfect.
I did 3 fresh installs with stock settings on 2.1 RC2:
1st: On second boot gets stuck at flower (after normal re-sizing and 15s reboot) I left it there for 5 minutes, re-plugging RPi2 will go past the flower, but will freeze on first or so frame. 4rd boot was weirdly stable, played a bit, was holding fine, on 5th it froze after 3 minutes.
2nd: Everything went perfect from resizing to first boot and rebooting... except the fact that it will freeze after 2 or so minutes just as I originally described.
3rd and last: exactly like the 2nd one.

Kivutar, 2017-06-23:
OK on my side I noticed that the only change I made on the RPi kernel was the RT patch. I forgot to add the performance governor.

parklez, 2017-06-24:
Sorry for not providing the log as instructed, here's from fresh install to the moment I used Putty to log it:

Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 882fe09)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4
Built: Jun 11 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 882fe09
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [Config]: Loading default config.
RetroArch [INFO] :: [Config]: loading config from: (null).
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch                                    .cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: [Audio]: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/slang-conversion-template.glsl"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: [Video]: Video @ fullscreen
RetroArch [INFO] :: [Video]: Starting threaded video driver ...
RetroArch [INFO] :: [EGL] Falling back to eglGetDisplay
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: [GL]: Found GL context: videocore
RetroArch [INFO] :: [GL]: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: [GL]: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
[... 4 lines cut ...]
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: Querying EGL extension: KHR_image => exists
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Video]: Graphics driver did not initialize an input driver.                                     Attempting to pick a suitable driver.
RetroArch [WARN] :: [udev]: Couldn't open any keyboard, mouse or touchpad. Are p                                    ermissions set correctly for /dev/input/event*?
RetroArch [INFO] :: [udev]: Plugged pad: Microntek              USB Joystick                                               (121:6) on port #0.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event0) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event0) supports 16 force feedbac                                    k effects.
RetroArch [INFO] :: [Autodetect]: 179 profiles found.
RetroArch [INFO] :: [Joypad]: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: [Autodetect]: selected configuration: /tmp/joypads/udev/Micr                                    ontek              USB Joystick          .cfg
RetroArch [INFO] :: [Menu]: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_hi                                    story.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_mu                                    sic_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_vi                                    deo_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_im                                    age_history.lpl].

Kivutar, 2017-06-24:
This new images is the latest master with the RT patch still applied but disabled from the kernel config: http://static.kivutar.me/Lakka-RPi2.arm-2.1-rc3.img.gz

I'm trying it on my side. Can you try it too please?

parklez, 2017-06-25:
Will do, tomorrow or so. thanks! :)

parklez, 2017-06-25:
holding like a boss, good job @Kivutar ! :)

Kivutar, 2017-06-25:
Ok, it was indeed the fully preempt that was causing the issue. @larskj any idea why?

closed 2017-06-25 by Kivutar

larskj, 2017-06-26:
Not really, sometimes the RT patch can trigger bugs in the application code that is otherwise dorment (or happens very rarely) with the normal scheduler. But for now the best we can do is to disable it for arm.