libretro/Lakka-LibreELEC #154: Replace "per-core" overrides and shaders with "per-system" overrides and shaders

kevinfishburne, 2017-06-22:
The existing functionality for core overrides is excellent with the exception of cores which emulate more than one system (Genesis Plus GX does SMS and Genesis, MAME does countless). For example there is no way to use one set of core overrides for SMS and another for Genesis because they both use the same core.

This is an issue because cores emulating multiple systems interpret the original hardware's framebuffer differently and have different resolutions, offsets, cropping/overscan, brightness/contrast/saturation, etc. which cannot be corrected by per-core overrides and per-core shader settings (image-adjustment, for example) since their granularity does not include the actual hardware being emulated, but the emulator itself.

The real-world use case scenario in which this is a serious problem is working with Mike Chi of the RetroTINK to create a custom Lakka image using 2048x240 hdmi timings and per-core custom viewport, core options and shader settings to accurately match original hardware on CRT televisions in 240p.

While this might be challenging to implement for MAME (unless the specific board is documented in the game information and thus could be queried), it should be fairly straightforward to implement using the algorithm which determines which roms belong to which system.

Kivutar, 2017-06-22:
You're right. I think you should move this feature request in RetroArch issue tracker because it requires a lot of work on this side.

closed 2017-06-22 by Kivutar

kevinfishburne, 2017-06-24:
Will do, thanks.

kevinfishburne, 2017-06-25:
I filed it and got this response, which is simultaneously irritating and promising as it contains a proposed workaround which you (Kivutar) may be able to reasonably pull off (or help me pull off):

https://github.com/libretro/RetroArch/issues/5100

So the workaround basically involves renaming and compiling duplicate cores so one may be assigned to one system and the other to a different system. The only problem I see is that there still appears to be no way other than using playlists to tell Retroarch which core (or duplicate core) should be used for a given content type. Thoughts on all this?

Kivutar, 2017-06-25:
You can also use the core info to tell which core is associated to which file extension