libretro/Lakka-LibreELEC #155: Qumox (ShanWan) Controller Wireless 2.4G USB Dongle not working

Demkan, 2017-06-22:
Hello, the joypad works fine in Windows and Retropie, but doesn't work in Lakka. Tested on the latest images Lakka-RPi2.arm-2.0 and Lakka-RPi2.arm-2.1-rc2. After connecting the USB Dongle to the Raspberry Pi 3 two messages appear.

> 100%: ShanWan     PC/PS3/Android Gamepad (9571/1397) not configured.
> 100%: Device disconnected from port #0 ShanWan     PC/PS3/Android Gamepad (9571/1397).

This is the controller https://www.amazon.co.uk/Wireless-Bluetooth-Remote-Controller-Windows/dp/B01LYVFXS2

The problem was already mentioned in the issue https://github.com/libretro/Lakka/issues/670

It is supposedly due to Lakka missing the module hid_betopff (source code at http://elixir.free-electrons.com/linux/latest/source/drivers/hid/hid-betopff.c)

Is it possible to add this module to Lakka? Thanks!

**dmesg**
[1.911574] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[2.011361] usb 1-1.4: New USB device found, idVendor=2563, idProduct=0575
[2.011369] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[2.011375] usb 1-1.4: Product: PC/PS3/Android Gamepad
[2.011379] usb 1-1.4: Manufacturer: ShanWan
[2.021237] input: ShanWan     PC/PS3/Android Gamepad as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:2563:0575.0001/input/input0
[2.021620] hid-generic 0003:2563:0575.0001: input,hidraw0: USB HID v1.11 Gamepad [ShanWan     PC/PS3/Android Gamepad] on usb-3f980000.usb-1.4/input0
[2.258579] systemd[1]: System time before build time, advancing clock.
[2.276417] systemd[1]: systemd 232 running in system mode. (-PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK -SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS -ACL -XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN)
[2.276719] systemd[1]: Detected architecture arm.
[2.276940] systemd[1]: Set hostname to <Lakka>.
[2.341102] systemd[1]: Listening on Journal Socket (/dev/log).
[2.341373] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[2.341480] systemd[1]: Listening on /dev/initctl Compatibility Named Pipe.
[2.342371] systemd[1]: Created slice System Slice.
[2.343746] systemd[1]: Mounting Variable Directory...
[2.345696] systemd[1]: Mounting POSIX Message Queue File System...
[2.391478] usb 1-1.4: USB disconnect, device number 4
[2.439253] systemd[1]: Created slice User and Session Slice.
[2.596849] systemd-journald[222]: Received request to flush runtime journal from PID 1
[2.644954] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[2.747998] usb 1-1.4: New USB device found, idVendor=20bc, idProduct=5500
[2.748007] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[2.748013] usb 1-1.4: Product: PC/PS3/Android Gamepad
[2.748016] usb 1-1.4: Manufacturer: ShanWan

gouchi, 2017-06-23:
Should be fixed with [this commit](https://github.com/libretro/Lakka-LibreELEC/commit/ed5687802c18b3b07c18daee9018976a0b305644).

Demkan, 2017-06-23:
Thank you gouchi for the fix. I'll test it and give a feedback when the 2.1-rc3 build comes out.

Demkan, 2017-07-01:
Just tested the new rc3 build, and it doesn't work either. Oh well, then I'll have to buy a controller compatible with lakka.

gouchi, 2017-07-01:
Is it possible to get some [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/) and `dmesg` when you plugged your controller because it should be working.

Thank you.

Demkan, 2017-07-01:
Thank you gouchi for trying figuring out the source of the problem. I'll post the log tomorrow because I don't have the old router at hand through which I connect to Raspberry Pi via SSH.

Demkan, 2017-07-02:
Here are the logs

**system logs**
```
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 882fe09)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4
Built: Jun 13 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 882fe09
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [Config]: Loading default config.
RetroArch [INFO] :: [Config]: loading config from: (null).
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: [Audio]: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/slang-conversion-template.glsl"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: [Video]: Video @ fullscreen
RetroArch [INFO] :: [Video]: Starting threaded video driver ...
RetroArch [INFO] :: [EGL] Falling back to eglGetDisplay
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: [GL]: Found GL context: videocore
RetroArch [INFO] :: [GL]: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: [GL]: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
[... 304 lines cut ...]
[    8.720596] Bluetooth: HCI socket layer initialized
[    8.720615] Bluetooth: L2CAP socket layer initialized
[    8.720668] Bluetooth: SCO socket layer initialized
[    8.726213] Bluetooth: HCI UART driver ver 2.3
[    8.726221] Bluetooth: HCI UART protocol H4 registered
[    8.726223] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    8.726339] Bluetooth: HCI UART protocol Broadcom registered
[   22.549560] random: crng init done
[  121.088204] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[  121.204730] usb 1-1.4: New USB device found, idVendor=2563, idProduct=0575
[  121.204746] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber            =0
[  121.204754] usb 1-1.4: Product: PC/PS3/Android Gamepad
[  121.204761] usb 1-1.4: Manufacturer: ShanWan
[  121.215787] input: ShanWan     PC/PS3/Android Gamepad as /devices/platform/so            c/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:2563:0575.0003/input/input2
[  121.216889] hid-generic 0003:2563:0575.0003: input,hidraw2: USB HID v1.11 Gam            epad [ShanWan     PC/PS3/Android Gamepad] on usb-3f980000.usb-1.4/input0
[  121.722146] usb 1-1.4: USB disconnect, device number 5
[  121.964836] usb 1-1.4: new full-speed USB device number 6 using dwc_otg
[  122.064571] usb 1-1.4: New USB device found, idVendor=20bc, idProduct=5500
[  122.064579] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber            =0
[  122.064582] usb 1-1.4: Product: PC/PS3/Android Gamepad
[  122.064588] usb 1-1.4: Manufacturer: ShanWan
[  122.074304] input: ShanWan     PC/PS3/Android Gamepad as /devices/platform/so            c/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:20BC:5500.0004/input/input3
[  122.074710] betop 0003:20BC:5500.0004: input,hidraw2: USB HID v1.11 Gamepad [            ShanWan     PC/PS3/Android Gamepad] on usb-3f980000.usb-1.4/input0
[  122.074744] betop 0003:20BC:5500.0004: Force feedback for betop devices by hu            angbo <huangbobupt@163.com>
```

gouchi, 2017-07-07:
You are right I didn't see 
`100%: ShanWan PC/PS3/Android Gamepad (9571/1397) not configured.`

If possible, it will be great if you can provide your [gamepad profile](http://www.lakka.tv/doc/Contributing-your-joypad-config/) so that we can include it for the next release.

Thank you.

closed 2017-07-07 by gouchi