libretro/Lakka-LibreELEC #156: Orange Pi Lite gets extremely hot

mstroh76, 2017-06-23:
With release Version 2.0 for Orange Pi Lite the H3 CPU reaches 82 °C and more on idle (Menu).
CPU Frequency is at 680 MHz at all 4 cores, but the temperature is way too high.
With Armbian in idle state the CPU has around 52 °C.
I tried to replace script.bin with armbian version. Makes no difference to heat production.

So far no damage is made to the board, but performance is maybe reduced.
I will try to find additional information (CPU freq during cpu load) as soon as possible.

mstroh76, 2017-06-23:
no heatsink:
no additional tests!

with heat sink:
current frequency 912000/1200000 switching
temp  73-76°C
current: 0,81-0,90 mA

with head sink and fan (active cooling):
current frequency  1200000
temp 44-47 °C
current:  0,79 mA

After activating active cooling cpu frequency stayes at 1,2 GHz. So with only heat sink (or no heatsink) CPU limit was reached, causing lower cpu performance and emulation problems.

Kivutar, 2017-06-23:
Or maybe we should switch to another governor. I remember that @Ntemis set the governor to 'performance' for H3.

mstroh76, 2017-06-24:
With ondemand menu temperature drops to 66 °C (480 Mhz CPU Frequency)
With N64 emulation running, cpu frequency jumps between 480 MHz and 1,2 GHz, temperature is stable at around 69 °C.

If CPU load on one core is present all core increase clock. If I start 'dd if=/dev/zero of=/dev/null' temperature rises  >72 °C, and the cores start switching beteen 1,2 GHz an 912 MHz because of termal protection.

ondemand governor will resolve the problem for menu and low usage emulation but not for emulation where at least one core is full needed.  

The cpu settings and max freq. seams to be not stable with lakka (maybe because of GPU usage) compared to armbian.  

I will  do more tests with armbian to combare conditions with one cpu at full load.

Kivutar, 2017-06-24:
Thank you for these informations. It will certainly help us figure out what's the best solution.

Ntemis, 2017-06-24:
Tranforming this soc into a game machine needs every juice we can squizze
out of it. Just add a heatsink as we all do.

Kivutar, 2017-06-24:
So that's the final word in the matter? We tell everybody to use a heatsink?

Note that on RPi we're not using the performance governor, it's set on 'on demand'.

mstroh76, 2017-06-25:
I did new tests with a new orange pi lite board and armbian. Board A und B use the same hardware (PSU, SD card ... ). 1 Core full load runing more than 30 min.

Board A: 0,54 A, 71°C
Board B: 0,43 A, 56°C

I looks like the first board has much higher power consumtion and temperature. I think month ago the Board B was not that bad - its maybe damaged somehow.   
Actually I started Lakka with board B - it has 65 °C after 40 min menu.

Ntemis, 2017-06-25:
@kivutar changing performance gov to on demand won't change much regarding
heat when system is on full load all you gain is a relatively cooler system
on idle. If you want this am all in. Throttling this weak CPU while gaming
is not ideal but we can test if you like

mstroh76, 2017-06-27:
IMHO :
Good emulation performance is the most importent needed feature!
Best solution maybe "ondemand" at menu "performance" at active emulation.
Adding CPU temperature at Information in menu will let the people get a feeling for  the system temperature (/sys/devices/virtual/thermal/thermal_zone0/temp).

Weak CPU? 20% faster then Pi 2 at a verfy low price. 
(My solution: keep performance setting and improve cooling: https://twitter.com/m_stroh/status/878360735977439232)

closed 2017-10-21 by Kivutar