libretro/Lakka-LibreELEC #157: S905 chip box can not connect multiple ps3 handle

lafpub, 2017-06-25:
I am using version 2.0 of LAKKA
I am using the S905 chip box
I use the ps3 handle to connect LAKKA, with a different receiver
I expect to be able to connect multiple Bluetooth handles
However, the result is only connected to a Bluetooth handle, ps3 four led lights will flash slowly. I have tested the raspberry pie 3b connection ps3 handle no problem, orange pie pc board (plus Bluetooth adapter) connected ps3 handle no problem

Kivutar, 2017-06-25:
And what about the wired mode? Can you connect 3 in wired mode?

lafpub, 2017-06-26:
Wired mode insert a few ps3 handle no problem
When wireless mode:
I have repeatedly tested today, the three ps3 handle can identify (sometimes not connected), in the S905 box, four led slow flashing, that is to say that each handle is not a separate signal identification. When using the wired mode on the S905 box, the ps key can be used, and the ps key is not available in wireless mode. Probably so, the problem should be less complicated, hope my test is helpful to you

closed 2020-08-18 by Ntemis