libretro/Lakka-LibreELEC #158: Audio problems in some 32x sega games.

marleyap, 2017-06-27:
I am having audio problems in some 32x sega games on the pico drive lakka OS 2.1 rc2, the audio of some games are showing hisses, flaws such as brutal unleashed 32x, virtua fighter 32x, blackthrone, metal head and emulator version that Use is picodrive 1.91cbc93b6, I use lakka x64 on the PC on which is an HP pavilion g4-2117br. Please correct the problem in these games, thank you.

gouchi, 2017-08-06:
Do you reproduce the issue with [Lakka 2.1 rc4 ](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1-rc4.img.gz)?

ToKe79, 2018-01-31:
Closing this issue, as no response received. If the issue still exists in the latest [stable](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv) releases, please feel free to re-open this issue. Thank you!

closed 2018-01-31 by ToKe79