libretro/Lakka-LibreELEC #160: Graphic problems in top gear 3000

marleyap, 2017-06-28:
Hello, I'm having graphics problems in top gear 3000, which is, glitchs in cars and scenarios.
I use the version Lakka OS 2.1 RC2 PC version, core emulator is Snes9x 1.54.117054ee, my machine is an HP pavilion g4-2017-br with AMD dual-core A6-4400m, 6gb ram. Please correct this problem, thank you!

Kivutar, 2017-07-01:
Can you try to switch either to
  * Lakka 2.1-rc3 that I published yesterday night and has an updated version of snes9x 
  * bsnes mercury for the core

marleyap, 2017-07-03:
I tested the newer versions of the Snes9X libretto Lakka OS 2.1 rc3, but the bugs in the scenarios and cars still happen, and have a detail the official version of Snes9x this bug does not happen, maybe it is programming error from core snes9x to retroarch, until The version for windows has this same problem, this has to be repaired as soon as possible. And the Bsnes Mercury the problem is worse !!!  Thank you very much in advance!

Kivutar, 2017-07-03:
I'm surprised because bsnes is known as a super accurate core.
You're using a no-intro rom?

marleyap, 2017-07-03:
I use bsnes balanced mercury, now I do not know if it's the emulator's own problem or dsp-4 firmware that I use to run the top gear 3000.

marleyap, 2017-07-03:
Use to rom top gear 3000 clean

Ntemis, 2017-07-04:
Can we have your gpu?

marleyap, 2017-07-04:
Not really a gpu is an apu, it is dual core cpu with radeon HD 7520g .... my notebook is an hp pavilion g4- 2117br, and all games that use the chips dsp1,2,3 and 4 are showing bugs in Core bsnes accurate, bsnes balanced and bsnes performance and also in the mercury versions. In snes9x only top gear 3000 is presenting graphical flaws and others that use the other variations of the dsp chips are working normal and being this bug in top gear 3000 only in the versions snes9x core retroarch, these bugs happen in both the windows, linux, but platforms I do not know if these problems happen in android and etc, but in the official version snes9x and higan / bsnes works perfect.

marleyap, 2017-07-04:
[image]
[image]
Graphical errors in all bsnes versions '' normal '' and bsnes mercury, except in the bsnes mednafen version

marleyap, 2017-07-04:
[image]

And this is the graphical errors that happen in the snes9x retroarch core ... and that's it, but finally I hope they solve those problems.

Kivutar, 2017-07-10:
I think we should move this bug report to the repositories of snes9x and bsnes. But I still can't understand why the bug happens on both emulators.

kunfuchicken, 2018-01-05:
Hi, all my other snes games are running through bsnes mercury accurate just fine, but top gear 3000 wont even load. I get the message "Failed to load Content." I've tried other cores, still nothing. I can't seem to figure out the problem. Any ideas?

Ntemis, 2018-02-11:
Then lets call @twinaphex to share some light 👍

ToKe79, 2019-07-02:
stale

closed 2019-07-02 by ToKe79