libretro/Lakka-LibreELEC #162: Dell OptiPlex 3010 hangs on boot until keyboard is plugged in

montag64, 2017-07-01:
First off, excellent work on this. What you all are doing is really incredible.

Something I've noticed ever since I started messing with Lakka late last year--for my locally installed Lakka version (all the way up to **2.0 RC3 June 13th build**) the OS won't boot until I plug in a keyboard. Then it boots. It is 100% consistent in that no keyboard = no boot and keyboard = boot. 

Hardware is a Dell OptiPlex 3010, bios version A17 booted in UEFI mode. Not dual booting. Booting from 500 GB SSHD. 

I'm used to grub and debian but this is a bit different. Where can I add verbose flags in some boot conf file somewhere in the boot loader so that I can give you guys more useful information?

montag64, 2017-07-01:
Also I already have SSH access, etc.

Kivutar, 2017-07-01:
Very interesting.

How many USB ports the laptop has? Have you tried other combinations, like plugging a mouse instead of a keyboard?

If you have SSH access when the screen is black, can you try to launch retroarch manually and get some logs? http://www.lakka.tv/doc/Troubleshooting-Lakka/

montag64, 2017-07-01:
It's a desktop (medium form factor). I have, and mouse doesn't do anything and the screen stays black. I do not have SSH access prior to seeing the flower unfortunately. That's why I was hoping to add a -v flag to the bootloader config

gouchi, 2017-07-01:
Can you try to boot with ? Just type in the boot prompt
`live ssh debugging`

Thank you.

montag64, 2017-07-01:
I'm not sure what you mean? I don't get a boot prompt or text of any kind on boot

montag64, 2017-07-01:
also live is not recognized as a command when I'm ssh'ed in

gouchi, 2017-07-01:
When you boot to usb, just enter 
` live ssh nosplash debugging`

> Start your Lakka Box with an USB keyboard. Hit tab while on the boot screen, this will let you edit the cmdline.

[Source](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc)

montag64, 2017-07-02:
I always boot from disk when I have this problem (where Lakka's installed) so I don't know how helpful this will be. I went ahead and created a usb installer and did what you asked and this is what was on screen:

Unique identifier for this client: c81f6640de64
mount: mounting UUID=2404-4447 on /flash failed: No such file or directory
mount: mounting UUID=2404-4447 on /flash failed: No such file or directory

Eventually it'll continue boot. Of course the keyboard is already plugged in from entering the command.

montag64, 2017-07-02:
I also tried editing /flash/syslinux.cfg. It's unclear to me how to force it into verbose mode. Nothing online about it. I removed quiet and replaced it with verbose, still black on boot, no text.                                                  

DEFAULT linux
PROMPT 0

LABEL linux
 KERNEL /KERNEL
 APPEND boot=LABEL=System disk=LABEL=Storage  quiet vt.global_cursor_default=0 loglevel=2

montag64, 2017-07-07:
For shits and giggles I plugged in a 10 key USB and the thing booted just fine. So it's looking for an HID device on boot that isn't a mouse for whatever reason. Still haven't figured out how to enable verbose boot

Kivutar, 2017-07-07:
To force verbose mode, you need to remove `quiet` from the line.

I'm thinking of a BIOS or EFI or bootloader issue.

montag64, 2017-11-04:
This is still an issue. If I remove 'quiet' from the line, I still do not get any logging or words on screen (black screen) and I still cannot connect via ssh to pull logs because at that point I don't believe the boot has gotten that far. 

If I load Windows in UEFI mode, it loads without issue and does not need a keyboard.

What can I do further to troubleshoot this?

gouchi, 2017-11-05:
It seems on both log we see 
```
input: Dell Dell USB Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3/2-1.3:1.0/0003:413C:2003.0001/input/input6
hid-generic 0003:413C:2003.0001: input,hidraw0: USB HID v1.10 Keyboard [Dell Dell USB Keyboard] on usb-0000:00:1d.0-1.3/input0
```

Can you try also to add `debugging` in the cmdline ?

Thank you.

ToKe79, 2018-01-31:
I never tried Lakka-Generic on a desktop, only on laptops - could this be a limitation of syslinux - that it will not display the `boot:` prompt unless there is a keyboard present? With laptop the keyboard is always present, or it is very rare to have desktop PC without an keyboard attached.

ToKe79, 2018-02-20:
Checked the logs, the only difference is, that in `with_keyboard` these 3 lines are extra:
```
[    2.642701] usb 2-1.4: new low-speed USB device number 4 using ehci-pci
[    2.747714] input: USB,2-axis 8-button gamepad   as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.0/0003:0583:2060.0002/input/input7
[    2.748096] hid-generic 0003:0583:2060.0002: input,hidraw1: USB HID v1.10 Joystick [USB,2-axis 8-button gamepad  ] on usb-0000:00:1d.0-1.4/input0

```
This seems that in both scenarios the keyboard was plugged in and instead of unplugging the keyboard, you have unplugged the game-pad. Or am I wrong?

As already @gouchi pointed out https://github.com/libretro/Lakka-LibreELEC/issues/162#issuecomment-341979297, the keyboard appears in both logs.

ToKe79, 2018-06-05:
Closing stale issue, feel free to reopen / comment if issue persists in the nightly builds (http://nightly.builds.lakka.tv/).

closed 2018-06-05 by ToKe79