libretro/Lakka-LibreELEC #163: Randomly losing assets and controller configs

crayonized, 2017-07-01:
Fresh install of Lakka 2.1 RC3. Tried a few cores, some of them crashed as usual resulting in my PC hard locking, as usual. After this had happened a couple of times, RetroArch can no longer find the assets or controller configs. So now I'm left with black boxes, a bitmap font and keyboard input.

Kivutar, 2017-07-01:
We need to understand why we get these filesystem corruptions. It's a major issue.

marleyap, 2017-07-02:
I'm having this same problem, this is a bit embarrassing. So please let this problem be corrected soon and if possible.

hooby3dfx, 2017-07-11:
I have also started noticing this issue since 2.0

Kivutar, 2017-07-11:
This problem has been there since the very beginning of Lakka sadly... I have no idea why the filesystem gets corrupted. It will also break the SSH service because systemd will be unable to open the sshd.conf, while this file is there on the disk.

Deleting the .config and .cache folder usually fixes it.

If anybody has an idea on how to find the cause of this bug, please tell us. I'm stuck on this.

crayonized, 2017-07-12:
I'm not a dev so take it for what it is but.. Maybe you can start with the method of exclusion. Does it happen on any other platform with RetroArch? Does it happen with LibreELEC or OpenELEC? If no and no, maybe look at the things you are doing differently from LibreELEC. Maybe compile and test a build that is as barebones are possible, and if the problem goes away, you can progressively enable things one by one. You might want to talk to the LibreELEC devs also.

Kivutar, 2017-07-12:
Yes it's a good idea. From what I understand, RetroArch keeps the retroarch.cfg open during the game. When the system is not shutdown properly or a game crashes, the file is never closed. I understand that it can cause problems bu I don't see how it can cause the whole filesystem to break.

I've been using RetroArch on my Arch Linux for years now and this problem never happened there.

crayonized, 2017-07-12:
Indeed I've also been using RetroArch on the desktop (Linux and Windows) since the SSNES days and never experienced filesystem corruption. I have no idea how things are with upstream OpenELEC or LibreELEC though.

ToKe79, 2018-02-20:
Is this still a thing? If yes, please reopen the issue.

closed 2018-02-20 by ToKe79