libretro/Lakka-LibreELEC #166: Stuck in flower but working live version

zetaXX, 2017-07-07:
Hello Im using Lakka 32 bits version in an old PC (ATI Radeon x300 graphic card), live version works just fine but installed version just stays stuck into the flower black screen and after a while I get a message in the left top corner: 
Error in mount_storage: mount_common: Could not mount LABEL=Storage
starting debugging shell... type exit to quit
sh: can't access tty: job control turned off

Kivutar, 2017-07-07:
OK, can you try to reinstall again? It can't find the Storage partition, that's not normal.

ToKe79, 2018-01-31:
Issue closed as no response received. If the issue still exists with latest [release](http://www.lakka.tv/get) and/or [nightly](http://nightly.builds.lakka.tv), please feel free to re-open this issue. Thank you.

closed 2018-01-31 by ToKe79

evbo, 2020-01-17:
I am seeing this issue again on `2.3.1`

I've reinstalled twice now. I suspect the flash drive I'm booting from may be getting corrupted by a faulty power supply, as the issue does not consistently occur.