libretro/Lakka-LibreELEC #169: Request: update Parallel core

tonyhoro, 2017-07-10:
Recent versions of parallel core have the much requested feature of mapping C-buttons as individual retropad buttons, instead of being stuck with right stick mapping.

It would be nice to have this on lakka as soon as possible.

gouchi, 2017-07-10:
If you are talking about [this commit](https://github.com/libretro/parallel-n64/commit/ec220576271d44eca22027a6aecc86fe29520bb1), it is already included in Lakka. Did you make a test with version [2.1 RC3](http://www.lakka.tv/articles/2017/07/01/lakka-21-rc3-with-citra-core) ? If it is not working, it will be available in the next release.

tonyhoro, 2017-07-11:
Thanks! I updated it and its working like charm. I wonder if this could be the default config on next releases.

closed 2017-07-11 by Kivutar