libretro/Lakka-LibreELEC #170: Dualshock 3 rumble/vibration not working

hooby3dfx, 2017-07-11:
- Which version of Lakka are you using?
2.0
- What system hardware are you using?
Orange Pi, USB Bluetooth adapter, Dualshock 3
- What did you do?
Play PS1 games using PCSX ReArmed.
I can confirm the rumble is working with other controllers. 
- What did you expect to happen?
Rumble should work on DS3.
- What happened instead?
Rumble does not work. The controller works alright otherwise.

Diskutant, 2017-08-29:
Same here. Couldn't get rumble to work yet with either DS3 nor DS4.
(Lakka 2.0 on a Pi3)

Kivutar, 2017-08-29:
We have `CONFIG_SONY_FF=y` in the kernel, so it must come from RetroArch..

Ntemis, 2018-02-14:
Since we switched to mainline kernel am closing this

closed 2018-02-14 by Ntemis