libretro/Lakka-LibreELEC #171: WiFi and Bluetooth problems on Cubietruck

batyrlan, 2017-07-12:
I've used Lakka-a20.arm-2.0 image for Cubietruck. After launching Retroarch I've tried to configure WiFi but there wasn't any WiFi AP in list. Also My DS3 did not connected to CT.

I suppose that this problem occurs due to wrong core module. For Linux-Sunxi's 3.4 kernel there should be AP6210 module. This module support WiFi and Bluetooth in Cubietruck. But in Lakka this module doesn't load during booting.

My apologies for my English, it's not my native language.

Kivutar, 2017-07-12:
Yes it is a known bug and we can't fix it easily.

A20 requires a super old kernel (3.4) and systemd droped support for kernels older than 3.6 or 3.10 I think.

Even if we add back the AP6210 module, it will be unusable.

batyrlan, 2017-07-12:
Hm, but uname shows that Lakka uses 3.4.104 kernel, but wifi still doesn't work. :-(

Kivutar, 2017-07-16:
Yes this is what I was explaining, 3.4.104 is too old for systemd

closed 2017-09-14 by Kivutar