libretro/Lakka-LibreELEC #176: stuck at flower screen

pinedaMario, 2017-07-15:
Hi guys, I need your help, this is the first time i'm going to use lakka but i'm unable to make it work, the detailes are as follows:

        Which version of Lakka are you using?
	Generic.x86_64- 
	2.0 from the web page
	2.1-RC3 from the latest nightly version

    What system hardware are you using? 
	https://www.cnet.com/products/compaq-presario-all-in-one-cq1-1406la-atom-d425-1-8-ghz-monitor-lcd-18-5-series/specs/

    What did you do?
	inserted the usb and tried live, it got stuck
	tried to install it to the HDD and it got stuck too

	i'm unable to access ssh over the lan but i got connection refused and ctrl+alt+f3 just dispay a blank screen

	after include the "tty retroarch=0" shown at the page http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/ at the usb live i get the installation menu every time

    What did you expect to happen?
	see the GUI after first boot 

    What happened instead?
	it got stucked at flower screen

i were unable to get the log files

gouchi, 2017-07-15:
Hi,

Thank you for you feedback. You can get ssh running if you boot with
`live ssh tty retroarch=0`

It will be useful if you can provide [Lakka system and graphic logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

pinedaMario, 2017-07-15:
Hi,

thanks for your answer, attached are the logs

[audioDeviceLogs.txt](https://github.com/libretro/Lakka-LibreELEC/files/1150678/audioDeviceLogs.txt)
[inputDeviceLog.txt](https://github.com/libretro/Lakka-LibreELEC/files/1150679/inputDeviceLog.txt)
[graphicCardLog.txt](https://github.com/libretro/Lakka-LibreELEC/files/1150672/graphicCardLog.txt)
[systemLogs.txt](https://github.com/libretro/Lakka-LibreELEC/files/1150674/systemLogs.txt)

gouchi, 2017-07-16:
Hi,

Thank you for the logs.

Can you try to edit [Lakka configuration file](http://www.lakka.tv/doc/Configuring-Lakka/) ? And put 

`video_monitor_index = "2"`

If possible also if you can provide the output of 

`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

Thank you.

pinedaMario, 2017-07-16:
Hi,

the configuration file change is done and the output for the command is:

LVDS-1: connected
VGA-1: connected

I modified the <<menu_shader_pipeline = "0">> according the FAQ because the performance was too bad.

It seems is working fine now, is there anything else i should do?

Thank you.

gouchi, 2017-07-16:
Glad it works with video_monitor_index.

You can also (e)nable, (d)isable output with 
`live ssh video=LVDS-1:d video=VGA-1:e `

Thank you for reading the FAQ ;-)

closed 2017-07-16 by gouchi