libretro/Lakka-LibreELEC #177: ScummVM: Error running game: Game data not found

geektbee, 2017-07-16:
Applications being run with ScummVM receive the following error message: "Error running game: Game data not found"
Manually specifying the path in the option area allows app to run.

Research indicates that there is a [new core available](https://github.com/libretro/scummvm/pull/72) as of 11 days ago.

Most recent version of Lakka that I've found for my system is Generic.x86_64-2.1-rc3, compiled more than 30 days ago; no nightlies after that are currently available for trying.

Requesting at least a new scummvm_libretro.so for us to copy over and/or a new full nightly of Lakka.

RobLoach, 2017-07-16:
Are you using a .scummvm file? What game are you launching? What does the directory structure look like? In the mean time, I'll put up a PR to update ScummVM in Lakka. Thanks!

geektbee, 2017-07-16:
Yes, I am using a .scummvm file in each data folder as described in the following links, as well as another resource or two I'm not in a position to find right now.

[Lakka Documentation - ScummVM](http://www.lakka.tv/doc/ScummVM/)
[Libretro Lakka-LibreELEC ScummVM wiki](https://github.com/libretro/Lakka-LibreELEC/wiki/ScummVM)

The file structure reflects those that's found in the wikis and docs.

```
/storage/roms/scummvm/fbear/
/storage/roms/scummvm/fbear/{a bunch of files}
/storage/roms/scummvm/fbear/fbear.scummvm

/storage/roms/scummvm/puttputt/
/storage/roms/scummvm/puttputt/{a bunch of files}
/storage/roms/scummvm/puttputt/puttputt.scummvm
```

Each of the *.scummvm files have the Unique ID code referred to in the docs. in this case, _puttputt_ and _fbear_. I've tried to modify the *.scummvm files with different CR and LF combinations, and lack thereof, but that also hasn't gotten me anywhere, either.

The scummvm.lpl is in the default location for playlists
```
/storage/playlists/scummvm.lpl
```

Sampling of contents:
```
/storage/roms/scummvm/fbear/fbear.scummvm
Fatty Bear's Birthday Surprise
/tmp/cores/scummvm_libretro.so
ScummVM
34AC3D85|crc
ScummVM.lpl
/storage/roms/scummvm/puttputt/puttputt.scummvm
Putt-Putt Joins the Parade
/tmp/cores/scummvm_libretro.so
ScummVM
27C3D661|crc
ScummVM.lpl
```

geektbee, 2017-07-26:
With the guidance of some of the folks over at the [libretro/scummvm github](https://github.com/libretro/scummvm/issues/73#issuecomment-317427440), I was able to successfully compile an updated scummvm_libretro.so file. I've used it on my "Generic" unit, and it now works as expected! Below is a link to the file that I'll keep for a few months, in case someone wants to use it or  upload it to the project.

[scummvm_libretro.so](https://www.dropbox.com/s/0gfhpnya1dgchnc/scummvm_libretro.so?dl=0)

Issue can be closed as resolved, as far as I'm concerned.

closed 2017-07-26 by geektbee

RobLoach, 2017-07-26:
Next release of Lakka will have the updated ScummVM sources. Thanks for testing! :+1: