libretro/Lakka-LibreELEC #179: Lakka-Lite: Core-less build

RobLoach, 2017-07-17:
Create a build of Lakka that doesn't have any Cores built into it. While downloading Cores through the online updater in Lakka can be problematic, having a core-less Lakka build may help in distribution.

Kivutar, 2017-07-17:
For people who want to build a core-less image, here's how to do it:

Edit the option file in distributions/Lakka, and empty the core list.

Kivutar, 2017-10-11:
Cores from the buildbot need to be tested on Lakka. This can be done by using the overlay system.

RobLoach, 2017-10-11:
Going to close this one, feel free to re-open if interested. Thanks Kiv!

RobLoach, 2018-01-24:
Re-purposing this issue to build LibreLakka, which would be Lakka, minus the non-commercial software. Removing the non-commercial cores and software would allow manufacturers to sell hardware with LibreLakka pre-installed.

It would be missing some features, but it is a step in the right direction. See the [libretro Licenses](https://docs.libretro.com/tech/licenses/) to see which software we would remove in this LibreLakka build.

ToKe79, 2018-01-24:
Altough I do not support this idea (the manufacturers are able to build their own images without the cores that cannot be distributed with sold hardware), but I think that it could be passed as a build argument e.g. `LIBREBUILD=true` and then modify if LIBREBUILD = "true":

- LIBRETRO_CORES in distributions/Lakka/options
- LIBRETRO_CORES in packages/Generic/package.mk (remove I think some of the additional cores for Generic project have non-commercial license, or move this to project/NAME/options and modify there; the remaining depend packages can remain in package.mk) + any modify any additional projects that somewhere add libretro cores (not aware of any others)
- IMAGE_NAME in scripts/image to include "Libre" in image - e.g. LibreLakka-Generic.x86_64-2.1

and also add to config/options
```bash
if [ -z "$LIBREBUILD" ] ; then
  LIBREBUILD=false
else
  LIBREBUILD="$LIBREBUILD"
fi
```
Any other ideas?

Ntemis, 2018-01-24:
Will be better to flag this as Nonfree=true/false

ToKe79, 2018-01-24:
As `true` being default? = build including "nonfree" cores and image without "Libre*"

bparker06, 2018-01-26:
To call it libre, wouldn't you also need to leave out the GPU blobs and any other non-free software not part of RetroArch?

markwkidd, 2018-01-26:
I think that it would be a great step forward for the emulation community to have an integrated emulation software environment to steer manufacturers/hardware dealers towards rather than just away from.

I'm definitely subscribing to this issue. :)

kivutar, 2018-01-27:
Ok. I agree with the env var idea of ToKe79.

I don't care much about the variable name, can be LIBRE or NONFREE or anything that is intuitive.

By default, it should build with the non-free software, because it is what we have been doing from the start.

Also, bparker's remark is right. But there are some parts that will always remain proprietary, for example:

 - the bootloader of RPi
 - GPU blobs like mali blobs

And certainly other packages that I forgot.

twinaphex, 2018-01-28:
I think this is a very bad idea. 

At the very least if we are to make this, I wouldn't make this image open in any way. You don't want even more parasitism and abuse from these 'free-market' libertarians whom honestly just want to make free easy money off this project and deprive the host of any and all resources while they prop themselves up.

Ntemis, 2018-01-28:
I fully agree with you. Is like giving them our hard work served in a hot plate to consume on our back with a frenic cashflow

twinaphex, 2018-01-28:
Yeah. Fortunately, vudiq has removed the link on the nightlies so far, so that crisis has been averted for now. We can think this through in a much more smart fashion, and probably somewhere on either Discord or IRC itself.

twinaphex, 2018-01-28:
Also, I have ordered the Lakka build servers now. E-mails were forwarded to kivutar, so hopefully Lakka will have fully working build servers shortly.

closed 2018-01-28 by twinaphex

TrashMacNugget, 2019-01-05:
@twinaphex As an end user; I try to have as little nonfree software in my life as possible. Not being able to do this without manually investigating and installing cores makes me less likely to use Lakka. What options are available for a regular user like me to use Lakka without any nonfree cores?

natinusala, 2019-01-05:
Yes, remove any core you don't want from there and rebuild the image. Be aware that depending on your platform you won't be able to have a 100% free image.