libretro/Lakka-LibreELEC #180: 20 sec of black screen

sorafire, 2017-07-17:
i am using mxqbox M201D lakka

it take 20 sec of black screen after lakka logo and then it load 
is that normal ?

Ntemis, 2017-07-20:
Depend on your sdcard performance and isnt that box 512mb ram only?

sorafire, 2017-07-20:
i dont think so it says on the box 1 gb and i am using sanDisk ultra with a 10 on a circle  and it have a speed up to 80MB/s

Ntemis, 2017-07-21:
ok so if you provide me with an up2date meson8b_m201d.dtb thats would
possibly fix your issues.
We are currently using this one
https://github.com/kszaq/LibreELEC.tv/blob/libreelec-8.2/projects/S805/patches/linux/meson8b_m201d_dtd.patch
Feel free to edit compile and test on you box(i don't have it) and sent us
a PR when fixed.
Regards

sorafire, 2017-07-21:
i am new at this can you help ?
how to add this to a new image ?

Ntemis, 2017-07-21:
Its added already but apparenlty needs fixing-> your job

sorafire, 2017-07-21:
how to add it with program you use ?

Ntemis, 2018-02-14:
Closing this as i have no intention to support every Chinese box ever created.
People should stick to something with Q&T

closed 2018-02-14 by Ntemis