libretro/Lakka-LibreELEC #181: wifi/bluetooth nolonger working with 2.x

Majoros, 2017-07-18:
**- Which version of Lakka are you using?**
    2.0 stable
**- What system hardware are you using?**
    x86_64
    MSI Intel Skylake H170 LGA 1151 DDR4 USB 3.1 Mini ITX Motherboard
    Intel 3.70 GHz Core i3-6100 3M Cache Processor
    Intel wireless 8260 rev 3a (m.2)
**- What did you do?**
   upgraded from 1.3.6. to 2.0. Bluetooth was working fine with 1.3.6 but stopped working with 2.0 ont he same hardware.
**- What did you expect to happen?**
   bluetooth to work
**- What happened instead?**
  bluetooth did not work

**plugging the ps3 controler in via USB...**
```
# dmesg|grep -i playstation
[  225.814803] input: Sony PLAYSTATION(R)3 Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:054C:0268.0005/input/input18
[  225.815328] sony 0003:054C:0268.0005: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-0000:00:14.0-1/input0
```

**dmesg failures and errors** 

```
Lakka:~ # dmesg|grep -i error
[    0.012982] ACPI Error: [\_SB_.PCI0.XHC_.RHUB.HS11] Namespace lookup failure, AE_NOT_FOUND (20160831/dswload-210)
[    0.017222] ACPI Error: 1 table load failures, 7 successful (20160831/tbxfload-246)
[    0.881262] acpi PNP0A08:00: _OSC failed (AE_ERROR); disabling ASPM
[    1.525515] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    1.525521] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    1.964653] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    1.964655] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    7.914331] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-26.ucode failed with error -2
[    7.914360] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-25.ucode failed with error -2
[    7.914388] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-24.ucode failed with error -2
[    7.914414] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-23.ucode failed with error -2
[    8.185366] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    8.185371] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    8.306490] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    8.306498] [drm:drm_edid_block_valid] *ERROR* EDID checksum is invalid, remainder is 107
[    8.486284] usb 1-11: device descriptor read/64, error -110
[   24.056678] usb 1-11: device descriptor read/64, error -110
[   29.603367] usb 1-11: device descriptor read/64, error -110
[   45.176782] usb 1-11: device descriptor read/64, error -110
[   51.023452] usb 1-11: device not accepting address 6, error -71
[   56.570135] usb 1-11: device not accepting address 7, error -71
Lakka:~ # dmesg|grep -i fail
[    0.012982] ACPI Error: [\_SB_.PCI0.XHC_.RHUB.HS11] Namespace lookup failure, AE_NOT_FOUND (20160831/dswload-210)
[    0.017222] ACPI Error: 1 table load failures, 7 successful (20160831/tbxfload-246)
[    0.881262] acpi PNP0A08:00: _OSC failed (AE_ERROR); disabling ASPM
[    7.914331] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-26.ucode failed with error -2
[    7.914360] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-25.ucode failed with error -2
[    7.914388] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-24.ucode failed with error -2
[    7.914414] iwlwifi 0000:03:00.0: Direct firmware load for iwlwifi-8000C-23.ucode failed with error -2
[    8.072743] thermal thermal_zone2: failed to read out thermal zone (-5)
```

To spite the errors it is attepting to load the correct drivers according to...
https://www.intel.com/content/www/us/en/support/network-and-i-o/wireless-networking/000005511.html

gouchi, 2017-08-19:
It seems those firmwares have been disabled upstream because [they don't work well](https://bugzilla.kernel.org/show_bug.cgi?id=196395#c9) so it is not available on [the repository](https://github.com/LibreELEC/iwlwifi-firmware/tree/master/firmware) we are using.

Majoros, 2017-08-19:
Well that disappointing. Thanks for looking into it.

Kivutar, 2017-08-20:
> Regarding the -22 firmware specifically, we had to remove that from all
> releases because it wasn't working at all for some hardware
> configurations. Unless those issues have been fixed we won't be able to
> pull in that firmware version.

I don't get it. If it was working for some configurations, why they didn't keep it?

gouchi, 2017-12-16:
Can you make a test if you reproduce your issue with [Lakka 2.1](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1.img.gz) ?

Thank you.

ToKe79, 2018-05-23:
No response received. Please re-open/comment in case not solved.

closed 2018-05-23 by ToKe79