libretro/Lakka-LibreELEC #182: Request: command line ROM scanner

skpy, 2017-07-19:
Just getting started with Lakka on a Raspberry Pi 3.  I have an NFS mount with many ROMs in sub-directories.  If I use the Lakka GUI to scan a directory, the GUI freezes while it enumerates the files on the mount point.  Then it takes a long time to actually scan each file.

To make things a little easier, I've organized the ROMs in directories by first letter (/storage/roms/nas/A, /storage/roms/nas/B, etc).  This reduces the lock-up time for parsing the directory, but means I need to go through all twenty six letters manually.

It would be ideal for me if I could ssh into the system and kick off the ROM scanning system from the command line.

Kivutar, 2017-07-21:
You can try this way https://github.com/libretro/Lakka-LibreELEC/wiki/Playlists

RobLoach, 2018-04-27:
https://github.com/kivutar/go-libretro-scan has potential.

closed 2020-08-18 by Ntemis