libretro/Lakka-LibreELEC #189: automatically write logs to a Samba share

markwkidd, 2017-08-01:
I was thinking about the troubleshooting process, and I was reading some of the threads where @gouchi helps new users gain SSH access in order to generate logs. It seems that if users didn't have to learn SSH and how to restart RetroArch in verbose mode, that would streamline troubleshooting quite a bit.

Therefore would it make sense for Lakka to save one complete set of logs somewhere that is accessible over Samba? I don't suggest an archive of logs that builds up over time, just logs from the current session, or perhaps the current session and the previous session.

After some discussion in the forums, gouchi suggested I create an Issue for this idea.

Forums user @escalade also suggested that one way to handle this is to log to /var/log in order to benefit from the "Logfile" SMB share which automatically creates tarballs from /var/log.

gouchi, 2017-08-06:
We may use [systemd-cat](https://www.freedesktop.org/software/systemd/man/systemd-cat.html) to do that.

gouchi, 2018-05-23:
It seems Kodi has this [features](https://github.com/LibreELEC/LibreELEC.tv/search?utf8=%E2%9C%93&q=createlog&type=) so we should adapt it to Lakka.

[Source](https://libreelec.wiki/how_to/provide_logfile#tab__via_sambasmb)

closed 2020-08-18 by Ntemis