libretro/Lakka-LibreELEC #194: RTL8188EUS Freezes

esfomeado, 2017-08-05:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.0
- What system hardware are you using?
RPi 2
TL-WN725N (RTL8188EUS) Wireless Adapter
- What did you do?
Tried to connect to a wireless network
- What did you expect to happen?
Connect successfully to the network,
- What happened instead?
Lakka froze.

**lsmod**
Module Size Used by
8021q 16074 0
cfg80211 229385 0
8188eu 688743 0
bcm2835_wdt 2777 0

**Lsusb -vvv**
https://pastebin.com/779VfN4z
bcm2835_gpiomem 2652 0

Tried to disabled power management but didn't solved the issue.

gouchi, 2017-08-05:
Can you try to make a test if your reproduce your issue with [Lakka 2.1 RC3 ](http://le.builds.lakka.tv/RPi2.arm/Lakka-RPi2.arm-2.1-rc3.img.gz)?

Thank you.

esfomeado, 2017-08-06:
@gouchi Works fine on on Lakka 2.1 RC3.

gouchi, 2017-08-06:
Good, thank you for the feedback !

closed 2017-08-06 by gouchi