libretro/Lakka-LibreELEC #196: ParaLLEI on Lakka 2.1 rc4

jdca, 2017-08-07:
It breaks when loading from states sometimes/often, it worked in 2.1 rc3 and some buttons on my pad have switched places for some reason?

Kivutar, 2017-10-11:
Can you check again with the newly released rc5 please?

Kivutar, 2017-10-11:
You are using a DS4?

jdca, 2017-10-12:
Yes ds4. I'll check it out on sunday or monday on both my rpi3 and my pc 64bit.

jdca, 2017-10-16:
I can connect my ds4 on Lakka-Generic.x86_64-2.1-rc5.img.gz using the bluetoothctl thru ssh via another computer as normal. 

But when I connect my ds4 after entering the Lakka UI my keyboard stop working and i can't really navigate because buttons can't be bind because when i press down on my digital d-pad it jumps 4 rows.

I managed too bind all buttons besides a and b after some not so smooth rageslammering on my keyboard but the ds4 won't work at all after this.

It kinda works on some buttons if I reset retroarch.conf and don't bind anything.

gouchi, 2018-02-14:
Hi,

Do you reproduce the issue with [Lakka 2.1](http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.1.img.gz) ?

For DS4 issue, please do as explained [here](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6).

Thank you.

ToKe79, 2018-06-05:
no response, closing

closed 2018-06-05 by ToKe79