libretro/Lakka-LibreELEC #199: NFS share not seen by RetroArch (but visible in CLI)

alegag, 2017-08-19:
Lakka version: 2.1-rc4
Hardware: ODROID XU4

If I mount NFS share from config file or from command line.  In both case, the mount is successful and the files are accessible from CLI.  I can read files, I can even copy a file from the share to the SDCard and play the game.  The mount point is in `/storage/roms`.

Under `/storage/roms`, there are folders, and those folders contains games.

But from the GUI, when I go in "Load Content" and in `/storage/roms`, I see "No items" message.  I don't even see the folders containing games (not only the games themselves, but really nothing from the share).  Calling the file directly from Playlist doesn't works either.

I go back to CLI, I still see and access those same files the GUI don't see.  I SSH as root and from what I can tell, the GUI runs as root too, so I suppose this is not a permission problem.

For what it worth, the owner of the NFS-share is "1000:101".  I've set the folders/files to "root:root" on the NAS to test (so that they also appear as "root:root" on Lakka), but same result.

It used to work flawlessly.  I've changed NAS-server in between and it stopped working (passing from NFSv3 to NFSv4.2).  The NFS server OS is Fedora.

Let me know what I can do to provide debbing information or any other useful data.

beh666, 2017-08-23:
I've noticed the same issue using an RPi3 and Lakka 2.1-rc1, 2, 3 and 4. Wrote a post about it a while back on the Lakka forum: https://forums.libretro.com/t/scanning-nfs-mounts-isnt-working/10772

Kivutar, 2017-08-30:
More informations from IRC:

> <bparker> kivutar: a strace log of RA would be helpful
> <bparker> to see what syscall is failing that's causing the dir list to be empty
> <TSCHAK_> dollars to donuts, it's a permissions mismatch... NFS4 is much more stringent regarding export permissions.
> <TSCHAK_> (in addition to requiring more running on the client, including an id mapper)
> <bparker> that makes no sense if the cli works though
> <bparker> I'm guessing RA has some flag set on a syscall that that new nfs version doesn't like
> <kivutar> TSCHAK_, RA runs under root here
> <TSCHAK_> yeah, that's meaningless on NFS4
> <TSCHAK_> deliberate design feature
> <kivutar> oh
> <TSCHAK_> NFS4 is a whole different beast to NFS3

bparker06, 2017-08-31:
>a strace log of RA

Hey, that was my comment.

alegag, 2017-08-31:
@Kivutar I'll compile RA under my Linux box and check if I can reproduce the bug connecting to the same NFS.

I have the same issue for CIFS by-the-way, slightly different tho', I'll open a different issue to avoid mixing things together.

Once I have RA running on my box, I'll send a strace if I'm able to reproduce the bug there.

If not, as strace is not provided on LibreELEC, let me know what I can do there to help with logtrace.

alegag, 2017-08-31:
@Kivutar On a Fedora Workstation, compiled the latest from RetroArch repository, it has access to the mounted NFSv4 share.

Either the bug as been fixed since the latest version of Lakka has been prepared, or it doesn't apply there (which I suspect is the case).

Let me know how I can provide you with useful information from LibreELEC !

beh666, 2017-09-02:
I don't believe (but i could be wrong) that this is related to NFS permissions.

I've tried the exact same mount options using Rasbian on my RPi3 and FreeBSD and Ubuntu on my x64 desktop PC. And I have no issues accessing NFS shares from inside RetroArch on these systems. I only get this problem when using Lakka.

alegag, 2017-09-02:
@beh666 This is also the conclusion I came to.

And the fact that it works from CLI is strange.

As reported in issue #202, I have a similar issue with Samba too.  I get load from "Set Content", but not from playlists.  And with Samba, I have no problem from CLI either.

So there is something funny going on in the LibreELEC+RetroArch mix somewhere !

@Kivutar Still open to help with debugging, just let me know what you need !

alegag, 2017-09-04:
If it can help.

From DOSBox, my mount failed, so I did a `mount c /storage`, I moved to `c:`.

In `c:`, I don't see the folder where I mount my share.  If I mount directly in `/storage/roms`, I don't see that directory from DOSBox, if I mount my network share to `/storage/roms/nas`, then, that's that directory I don't see from DOSBox (but I would see other native directory showing up).

alegag, 2017-09-06:
@Kivutar If it can be of any help, I can confirm the problem is not related to LibreELEC as I can see the NFS share from a LibreElec 8.0.4 in Kodi (mounted the same way I did on Lakka).

beh666, 2017-09-07:
Here are two theories I'll check myself whenever I have the time available: Is RetroArch running in its own separate sandbox? Or its own user, which isn't a member of the group needed to be able to access nfs shares?

alegag, 2017-09-10:
@Kivutar New information.

I've compiled LibreELEC for board and added RetroArch in the image.

I mount the NFS share, I see it from CLI as usual.

I also see the content from Kodi.

When I launch RetroArch, I don't see any files in the `roms` folder (where I did my mount point).

As reported in issue #202, if I try the LiveUSB version of Laka on a X86, it works fine.

So it really seems to be limited to the ARM implementation of RetroArch after-all.

Should I open a ticket overthere ?

beh666, 2017-09-11:
Hmm, does the LiveUSB image use FAT32 as a filesystem? In that case, it won't support file permissions and perhaps that's why NFS shares work?

alegag, 2017-09-23:
Any updates on this ?  I still can't use Lakka on NFS/CIFS.

Kivutar, 2017-09-23:
Some informations:
RetroArch is not running in a sandbox or a container
RetroArch run as root user
The partition is ext4

alegag, 2017-10-25:
Do we have some progress on that issue ?  Have this been reproducable by the RetroArch folks ?

beh666, 2019-09-14:
I don't have this problem anymore on my brand new RPi 3B+using the latest Lakka mounting an NFS share by following the instructions: https://www.lakka.tv/doc/serving-roms-from-a-nas/

No idea about CIFS/Samba as I've never used it.

ToKe79, 2019-09-14:
Thank you for your time and feedback.

closed 2019-09-14 by ToKe79