libretro/Lakka-LibreELEC #202: CIFS share partitially working

alegag, 2017-08-31:
Lakka version: 2.1-rc4
Hardware: ODROID XU4

Probably related to issue #199.

Samba version 4.6.7 running on a Fedora server with SELinux.

The shares works as expected from other systems (and even from a printer).

Share is accessible from CLI and works as expected.

In RA, from "Set Content" menu, I can browse, load a game and it plays just fine.

When I close the game and go to "Recent History", the same game that I just close won't open, the file is not found.

Any game in any playlist won't load either, even tho' the path is correct, RA complains that the file doesn't eixsts.

alegag, 2017-09-05:
This bug is not limited to Samba/Linux.

I tested a share to a Windows 10 box and had the same problem.

I can easily see the behavior by mounting the share to `/storage/roms`, starting DOSBox, doing `mount c /storage`, then doing `c:` and `dir`, and the folder `roms` don't appears in the directory listing.

So this seems to be a generic problem with network share (including the one in ticket #199).

@Kivutar If I can do anything to help debug this, let me know !

alegag, 2017-09-05:
Oh ! And to trow it here too, I've tested with RetroArch on a Fedora Workstation and the shares works just fine, so it really seems to be Lakka-specific.

Kivutar, 2017-09-05:
Thank you for this precious information. Have you tested the retroarch extensions on the officiel LibreELEC images to see if they are affected too? If they are not, we may be able to find a fix quicly.

alegag, 2017-09-05:
@Kivutar I can try for sure! I wasn't aware of this extension to LE, can you point me in the right direction so I can test this quicker?

Kivutar, 2017-09-05:
I don't follow the retroarch addons on LE, so I'm not sure. Maybe this one? https://forum.libreelec.tv/thread/287-gamestarter-retrogaming-add-ons-rpi-generic/

RobLoach, 2017-09-05:
I use Escalade's LE8 "Remix" a lot:
- https://forum.libreelec.tv/thread/302-le8-remix-generic-rpi-emulationstation-chrome-spotify-provide-logs-if-problems/
- https://github.com/escalade/LibreELEC.tv

alegag, 2017-09-06:
@Kivutar This option works only for RPi, I'm running a Odroid bord.

Same for @RobLoach suggestion, those are only supporting RPi.

So I can't test any further in this scenario.

What I can confirm tho' is that a fresh install of LibreElec 8.0.4 running Kodi, I can see the NFS share just fine (reference to ticket #199).

RobLoach, 2017-09-06:
> Same for @RobLoach suggestion, those are only supporting RPi.

Naw, Escalade's LE8 Remix works on Generic. Stick it on a SD card, run the install, and you're up and running. Downloads are over at http://bit.ly/2mVGdBf

alegag, 2017-09-08:
Okay, LE8 Remix won't compile for Odroid and Escalade said the arch is not supported in his fork, so I really can't test any further down this road.

@Kivutar Any other idea of something I could do to retreive more information ?

Kivutar, 2017-09-09:
@alegag maybe tyring a live USB on a PC?

alegag, 2017-09-10:
@Kivutar I tried from a live USB "Generic" with Lakka 2.0 (which didn't work either for me) and it worked for CIFS and NFS.

Ntemis, 2017-10-24:
This not a bug.
@alegag  read related info here https://libreelec.tv/2017/09/libreelec-krypton-v8-1-2-beta/

closed 2017-10-24 by Ntemis

Kivutar, 2017-10-25:
So what does he need to do? Configure his NAS to use Samba 2 or Samba 3 ?

Kivutar, 2017-10-25:
Or this?

> Kodi “Windows network browsing” needs the Kodi SMB client to support SMB1 and we now default to SMB2/3 connections so browsing no longer works. Windows SMB shares must be added using the full server/share path. SMB sources from Samba/NAS devices can normally be browsed via the “ZeroConf network browsing” alternative.

alegag, 2017-10-25:
@Ntemis This is a bug, please read the full issue.  I suppose that conclusion came from issue #242 ?  This is not the same thing.

The link you provided says the support for SMB 1&2 has been dropped, that SMB3&4 are supported and that it default to SMB4.

My Samba server is SMB4, so this is not the problem.

Furthermore, I have no problem on LibreELEC side. I'm able to mount the share and access all files from the CLI.

I only can't access files from RetroArch, so this is not an issue with LibreELEC directly.

Please, reopen this issue as this has not been solved and I still can't access any of my share from RetroArch.

Ntemis, 2017-10-25:
I dont have an odroid XU4 but kivutar can verified if indeed there is a bug here.

alegag, 2017-10-25:
@Ntemis As I wrote in my last comment, you can reproduce this on a RPi3 too, and with a share mount to a Windows 10 system.

alegag, 2017-11-30:
@Ntemis @Kivutar Please, reopen this issue as this as never been solved and still a problem on all ARM devices mounting a Windows share, including a share to Windows 10 (not only Samba).

All steps to reproduce have been listed above.

It still is a problem in 2.1 ARM (RPi3 or Odroid).

Thank you !

vevsvevs, 2018-11-10:
Just in case if someone faced the same issue and still trying to find a solution. In my environment (CoreElec + Samba v1 share mounted with systemd) option "noserverino" in .mount file solved the problem.