libretro/Lakka-LibreELEC #203: genesis_plus_gx for ARM chips non-RPi

zach-morris, 2017-08-31:
I was having an issue with genesis_plus_gx core after installing Lakka-LibreELEC on an S905 device.  I found that after loading a game, the system would freeze particularly when the screen has to scroll with the game.

After some digging, it looks like the core needs to be compiled with -DALIGN_LONG and -DALIGN_DWORD to resolve this issue.  Found this info [here](https://forums.libretro.com/t/genesis-plus-gx/41/4?u=zachmorris).  Some more info [here](https://github.com/ekeeke/Genesis-Plus-GX/issues/158).

 I assume the issue would be similar on ARM chip boxes that aren't RPi, like the ODROID, although I can't confirm since I don't have one.

Kivutar, 2017-09-05:
Thank you, I hope it will fix this problem (that I never noticed myself)

closed 2017-09-05 by Kivutar

zach-morris, 2017-09-16:
Just wanted to confirm this fixed the issue.  Thanks!

commit 57fab6dc (2017-09-05): Set -DALIGN_LONG for genesis-plus-gx on ARM. Should fix #203