libretro/Lakka-LibreELEC #213: Strange delays when starting retroarch or any core

TheShich, 2017-09-09:
[Delay.txt](https://github.com/libretro/Lakka-LibreELEC/files/1290055/Delay.txt)

- Which version of Lakka are you using? 2.0 live USB
- What system hardware are you using? PC Pentium4, 2Gb RAM, ATI Radeon 9600
- What did you do? Start retroarch or any core
- What did you expect to happen? fast start of retroarch and cores
- What happened instead? strange 10-15 seconds delays (see log file)

I noticed strange delays when using Lakka on my PC with all cores. It's strange, but disconnecting network cable removes this delays. Further investigation shows that when I starting from console like "retroarch -v", it fixes delays too (even with network enabled).
Moreover, 
1) Setting environment variable to "export DISPLAY=:0.0" (exactly like written inside retroarch.service file) and starting retroarch from console leads to delays.
2) Setting environment variable to "export DISPLAY=0:0.0" or unsetting DISPLAY variable removes delays.
I'm certain that DISPLAY variable in ":0.0" position causing delays for me with retroarch as service.
 I've tried to edit "retroarch.service" but it's look like the file located on squish filesystem.
It would be nice to get rid of these nasty delays and still have network enabled and have retroarch booting automatically without typing from keyboard every time.

Kivutar, 2017-09-14:
Thank you for this useful informations. Here is an image without the DISPLAY=:0.0

http://static.kivutar.me/Lakka-Generic.x86_64-2.1-rc5.img.gz

I don't know what this variable does, I just copied it from the KODI service. I will have to test if removing it causes regressions elsewhere.

TheShich, 2017-09-14:
DISPLAY is a variable used by X Window system (do Lakka use one?). It it used to choose where to display graphics output. The strange thing is that :0.0 is a valid and commonly used value meaning display 0, screen 0 on localhost. Moreover, they say that if hostname is not given, the most efficient way of communicating to a server on the same machine will be used. It seems to be not true in this case.
https://askubuntu.com/questions/432255/what-is-display-environment-variable

Kivutar, 2017-09-15:
We don't use xorg, so it should be OK to just remove the variable.

closed 2017-09-30 by Kivutar