libretro/Lakka-LibreELEC #214: Newer MAME cores

masid101, 2017-09-10:
Not really an issue, but is there any reason why the newer MAME cores like 2010 or 2014 are not bundled with Lakka? At least the x64/x86 builds should be able to handle them.

Kivutar, 2017-09-14:
mame2010 and mame2014 build fine for PC yes. They are super heavy however, something like 100mo for the binary. That's why I don't include them. But the packages definitions are there and I build them from time to time to check if they still build.

masid101, 2017-09-15:
Do 200 megs more or less for the lakka download really have any relevance today?

Kivutar, 2017-09-16:
Yeah, each image is now around 200M, and each batch of image is more than 10G.
If each image becomes 400M, we double this.
Our servers hard drives are full, so it affects the number of old builds we can provide for the public.
Also everybody is complaining about the slow downloads.
Also, where I'm located now, it can take a few hours for me to download a lakka image from the build server. So this is going to affect the number of images I can test per day.

masid101, 2017-09-17:
Okay, thanks for the info!

closed 2017-09-17 by masid101

Kivutar, 2017-09-17:
We may switch to a new organization of the cores, where we allow again to download cores from the buildbot, once we fix the crashes.
Meanwhile, you can download the core you want manually, and place it in the cores directory in Lakka, and it may work.

masid101, 2017-09-17:
Where can I find the cores for the manual download?

Kivutar, 2017-09-17:
There https://buildbot.libretro.com/nightly/linux/x86_64/latest/

masid101, 2017-09-17:
Thank you!

Maybe it is possible to bundle at least the playlist symbols for the newer cores. Then I can download the core manually and still have the nice looking frontend!

Kivutar, 2017-09-17:
The databases are there, so it should work

masid101, 2017-09-17:
No luck with MAME2014 and MAME2016 playlists here. Only black squares on 2.1 RC4. Maybe the same problem as here ... https://github.com/libretro/Lakka-LibreELEC/issues/201