libretro/Lakka-LibreELEC #215: Filesystem corruption has been detected

TheShich, 2017-09-11:
- Which version of Lakka are you using? Live USB 2.0 Lakka-Generic.i386-2.0
- What system hardware are you using? PC Athlon XP 2200+, 768M, NVIDIA GeForce FX 5600 XT 128M
- What did you do? Tried to startup Lakka
- What did you expect to happen? Regular boot 
- What happened instead? I saw following error:
Filesystem corruption has been detected
To prevent an automatic repair attempt continuing
press any key or power off your system within the next 120 seconds
The flash drive is 100% working, I tested it on another PC

Sobient, 2017-09-12:
This is more than likely a drive issue. Lakka doesn't corrupt external media. 
Remember to check for SD-card for corruption, try one of these tools: https://www.raymond.cc/blog/how-to-check-and-test-usb-flash-drive/

Additionally, you should post the specs of your flash drive, the capacity, block size the drive was formatted, along with its capable transfer speeds (class) and brand/age.

RobLoach, 2017-09-12:
> You should anyways check your SD-card for corruption

This. Also make sure the .img downloaded in full. Check the md5.

Kivutar, 2017-09-14:
There are in fact filesystem corruptions on the storage partition sometimes.

I've never been able to determine the cause, because it happens rarely, and when it does, debugging becomes difficult.

TheShich, 2017-09-14:
Dear friends
I, healthy in mind and body, declare that my 100% working brand new Leef Surge 8GB flash drive with 100% working and md5 checked Lakka image (I checked both Lakka-Generic.i386-2.0.img.gz and Lakka-Generic.i386-2.1-rc4.img.gz) written with Win32DiskImager.exe under Windows is not booting on above mentioned PC with following message:
"Filesystem corruption has been detected
To prevent an automatic repair attempt continuing"
No flash drive corruption occurs in my case. The message is fake. I can insert same flash drive into another PC and it will be booted perfectly. It looks like incompatibility of Lakka and this old PC with wrong message about corruption. The PC is able to boot other USB images, but not Lakka.
How can I investigate this issue further?

Kivutar, 2017-09-15:
Try to boot this image: http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20161221183017-r21729-g274b40f.img.gz

And then do a manual upgrade to the latest Lakka, by placing the tar in the update folder

Sobient, 2017-09-15:
@TheShich Have you tried other external drives? There might be incompabilities with your drive.

Kivutar, 2017-10-11:
No replies for a month, feel free to reopen.

closed 2017-10-11 by Kivutar

DutchAirRaidSiren, 2017-10-11:
I have the same issue. I also get the "filesystem corruption detected" error when I try to install Lakka on a PC. I checked if my USB drive (3 weeks old now, I believe.) was corrupted, however, this was not the case. The HDD also doesn't seem to be the problem either. Any other Linux OS will install on it. The weird this is that if I download a older version of Lakka, then I will be stuck on the cloudberry screen. I am fearing that my PC may be too old, but I don't think it is.
The PC that I want to install it on is a Dell Dimension 2200.
Specs:
512MB RAM.
Intel Celeron 1.3 GHz processor.
Intel 810E chipset.

I've tried so many other Linux emulator distros, but without success. The only one that worked was AGLAD, but I don't like that one very much. I would love to be able to play my favourite games with Lakka, and I've searched for this issue on internet multiple times, but it didn't fix anything.
A reply would be greatly appreciated.

phoenixweiss, 2017-12-08:
I have the same issue on Raspberry Pi 3B

DerekRoth, 2024-10-18:
Since this issue is the first Google result for the Lakka "Filesystem corruption has been detected" search, and i just had the issue with the latest version on a Raspberry Pi 5, here is the solution:

- log into your device with SSH (login root, password root)
- execute the following command: `resize2fs /dev/mmcblk0p2`
- reboot