libretro/Lakka-LibreELEC #216: Lakka and ODROID-C2

AtWT2014, 2017-09-14:
I used latest version 2.0 of Lakka for ODROID-C2

It has limited list of monitor resolution supported in boot.ini
I tried a resolution out of this list with no luck.
In particular, 1280x1024 doesn't supported.

Moreover, Lakka for ODROID-C2 has no support of latest C2 distros' "display_autodetect" feature which uses edid HDMI command during startup for automatic resolution detection.
See http://odroid.com/dokuwiki/doku.php?id=en:c2_auto_detect_display.

ODROID team member told me at forum that Lakka has old kernel and boot.ini
Please, merge latest ODROID system's changes with Lakka if and when it possible.

Thanks.

Kivutar, 2017-09-15:
OK thanks for the report. I will work on this after 2.1-RC5 is out.

Kivutar, 2017-10-17:
For now I prefer to follow LibreELEC 8.2 which is using the generic amlogic 3.14 kernel.

I don't want to diverge too much from upstream.

However, we can redeclare this bug in their tracker so the changes will be pulled to Lakka the next time we merge upstream.

closed 2017-10-17 by Kivutar

AtWT2014, 2017-10-17:
Thank you!

Would you be so kind to give me the reference to your ticket in their
tracker? Wanna get a finger on the pulse ))

With regards,
    Andrew

Kivutar, 2017-10-17:
I don't find their public tracker, so I contacted them on Slack for now.