libretro/Lakka-LibreELEC #219: Input issue when switching advanced/simple mode

Kivutar, 2017-09-15:
Tested on 2.1-rc5
Can be tested only when compiling RA with HAVE_LAKKA=1
This is a regression due to the latest optimizations on XMB

Go to Menu Settings
Enable Advanced Settings
Go back to the Main Menu
Go to Menu Settings again
Disable Advanced Settings
Hit Cancel
Press Right or Left

gouchi, 2017-09-16:
With commit [e5fdc50](https://github.com/libretro/RetroArch/tree/e5fdc50812b951f1e15ce8357d41c52ba033cd90), I got this [issue](https://imgur.com/a/gsjeJ). I can go up to get back the menu.

ToKe79, 2017-12-09:
IIRC I have this same also without HAVE_LAKKA=1 - I compile the RetroArch addon for LibreELEC/KODI without HAVE_LAKKA and have the same issue when switching off advanced settings..

Kivutar, 2017-12-10:
We need to add a `if` in `menu cbs left` and` menu cbs right` to handle the edge case where the navigation pointer is >= the number of menu entries.

closed 2020-08-18 by Ntemis