libretro/Lakka-LibreELEC #221: Remotely save/load/backup to/from Dropbox, AWS, anything

RogueRonin, 2017-09-18:
There was an issue on the old repo about this.
Wasn't mine, but couldn't find it here so I'll open it again.

I'd love to have all my savegames available from anywhere and have them backed-up safely and automatically. Then you could also load/save them with any other emulator like Dolphin or VBA, I guess.

metchebe, 2017-10-12:
You can create and schedule a script that makes backups with [rclone](https://rclone.org/). I use the ARM build on Lakka RPi2.

RogueRonin, 2017-10-29:
I was thinking about creating an in-built solution so that you can put your AWS,Dropbox credentials in the settings menu along with some folder name configuration or whatever and have everything done for you.
But this could use rclone or something similar under the hood.

closed 2020-08-18 by Ntemis