libretro/Lakka-LibreELEC #222: Add dispmanx support to Lakka

kevinfishburne, 2017-09-24:
https://www.raspberrypi.org/forums/viewtopic.php?t=25146

My friend Bahn Yuki recently tested RetroPie with dispmanx and it seems to reduce latency on the RPi 3 significantly (he said by two frames). Would it be possible to include this as a video driver in RetroArch, and/or include the dispmanx library in OpenELEC so RetroArch will present it as an option if already compiled to support it? We've made incredible progress on 240p gaming on CRT TVs and latency is one of the few remaining issues other than per-core hdmi timing switching (another issue which Lakka should incorporate that RetroPie allows). As always, thanks, and keep up the great work.

starquake, 2017-10-25:
Proof that this seems to be the case: https://forums.libretro.com/t/an-input-lag-investigation/4407/482

Kivutar, 2017-10-27:
Interesting.

Adding dispmanx support is certainly easy. I just need to check that it doesn't take precedence over the traditionnal vc+gles path that is required for 3D cores.

closed 2020-08-18 by Ntemis

starquake, 2020-08-19:
Any information on why this has been closed?