libretro/Lakka-LibreELEC #224: ODROID XU4 cannot boot with eMMC

hutzero, 2017-09-26:
I have tried on nightly build already. It still not fix. 
May this souce below could help you to find root of cause.

https://github.com/nadenislamarre/batocera.linux/tree/master/board/recalbox/xu4

Kivutar, 2017-09-28:
In fact we know the cause. The adapter eMMC2SD prevents us to access the first sector of the eMMC where we're supposed to flash u-boot.

Luckily, the eMMC you buy from hardkernel already have an u-boot flashed on the first sector. So we could generate a special image with an offset of 1, and it would then work for the eMMC (if it comes from hardkernel).

But it would stop working for SD.

hutzero, 2017-09-30:
Yes, I brought it from hardkernel. I can use batocera(recalbox) with both of eMMC and microSD in the same image. But if you cannot build Lakka in the same image like them, please split it to 2 versions. (I was ever try the previous special build from you in the forum, but not working)

Kivutar, 2017-10-11:
If they did it, then we can also do it. In fact, it was already the case on old versions of Lakka, before the LibreELEC switch. I should look into that.

Kivutar, 2017-10-12:
Hardkernel is going to send me two new eMMC and I will update their kernel and u-boot for XU3/4

hutzero, 2017-10-14:
Thank you very much for your hard work in this issue.  I will become a Patreon to support every month, if this issue has fixed. I promise.

Kivutar, 2017-10-16:
I found a fix