libretro/Lakka-LibreELEC #225: Retroarch segfault when removing game from playlist

ToKe79, 2017-09-27:
Version: 2.1-RC4
System: Generic.x86_64 (PC)
Reproduction: Select game from playlist, choose "Remove" (must allow in Settings -> Playlists -> Allow to remove entries = ON)
Expected behaviour: game removed from playlist
What happens: Retroarch segfaults (from dmesg):
retroarch[546]: segfault at 20 ip 00000000004bf5dc sp 00007ffdc0ce1620 error 4 in retroarch[400000+2d8000]

log.txt:
`RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 882fe09)
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 
Built: Jul 25 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 882fe09
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [Config]: Loading default config.
RetroArch [INFO] :: [Config]: loading config from: (null).
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: [Audio]: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/slang-conversion-template.glsl"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: [Video]: Video @ fullscreen
RetroArch [INFO] :: [Video]: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: no
RetroArch [INFO] :: [DRM]: Connector 0 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 10 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x854) 1280 x 854, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1280x720) 1280 x 720, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1152x768) 1152 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (848x480) 848 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (720x480) 720 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (640x480) 640 x 480, 59 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [GL]: Found GL context: kms
RetroArch [INFO] :: [GL]: Detecting screen resolution 1440x900.
RetroArch [INFO] :: [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7fabfc0304a0.
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 5760).
RetroArch [INFO] :: [GL]: Vendor: X.Org R300 Project, Renderer: Gallium 0.4 on ATI RV515.
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 13.0.6.
RetroArch [INFO] :: [GL]: Using resolution 1440x900
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: [GLSL]: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event4 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_TOUCHPAD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_TOUCHPAD.
RetroArch [ERROR] :: [udev] Failed to open device: /dev/input/mouse0 (Inappropriate ioctl for device).
RetroArch [INFO] :: [Joypad]: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [INFO] :: [Menu]: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
function is no-op
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 5760).
RetroArch [INFO] :: [KMS]: New FB: 1440x900 (stride: 5760).
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
Segmentation fault (core dumped)
`

ToKe79, 2017-09-29:
could not reproduce on RC5 @Kivutar provided today (RC5 build).

1985a, 2017-10-05:
I have the same problem when I tried to play any game on MAME core.

ToKe79, 2017-12-09:
bug gone - no  more segfault after removing item from playlist

closed 2017-12-09 by ToKe79