libretro/Lakka-LibreELEC #227: Could not mount /flash/SYSTEM on reboot after lakka-upgrade

toloselatrack, 2017-09-29:
- Which version of Lakka are you using? lakka 2.0 stable realese
- What system hardware are you using? raspberri pi3
- What did you do? lakka-upgrade at the shell in ssh (it seems it downloaded lakka 2.1rc3 image)
- What did you expect to happen? have an upgraded system ;-)
- What happened instead? the following boot error:
*** Error in check_update: mount_common: Could not mount /flash/SYSTEM ***
sh can't access tty: job control turned off

check screenshot here: https://imgur.com/a/b8mzg (SONY BRAVIA tv =)

Kivutar, 2017-09-29:
Ok, I think something went wrong during the upgrade.

To fix your installation, you can download this http://le.builds.lakka.tv/RPi2.arm/Lakka-RPi2.arm-2.1-rc4.tar and move the SYSTEM and kernel.img files on the first partition of your SD card

Kivutar, 2017-10-11:
I think it's an isolated issue, I cant' reproduce it.

closed 2017-10-11 by Kivutar

evbo, 2020-01-20:
I think this issue has repeated a few times in various forms: 
https://github.com/libretro/Lakka-LibreELEC/issues/166
https://github.com/libretro/Lakka-LibreELEC/issues/811

I think one very important piece of documentation that everyone debugging issues like these is that flash drives are highly susceptible to damage if a power source/supply is unclean. For instance, if you have a habit of turning off your computer and then removing the power cord, note that depending on your power supply, when you go to plug it back in the jolt of electricity can be very damaging in certain cases. Even if you have a power strip, flipping the switch on it isn't ideal either as the switch may "bounce", causing jitters in the signal delivered to your computer.

The bottom line is it's absolutely possible to corrupt a flash drive, especially when it is left plugged in for repeated shutdown/power on scenarios where all sorts of side affects may occur (especially if your home outlets aren't strictly following code!)

Recommendations: 

- hopefully your power supply has a toggle switch, such that it is debouncing when switched from off to on. Smaller power supplies sometimes don't have a switch, which means as soon as they are plugged in they are susceptible to unclean power.

- Consider purchasing an "uninterrupted Power Supply", which will help deliver "clean" power to your computer and thus flash drive

- At the very least, consider purchasing a very cheap USB hub with power switches. While this won't protect your flash drive completely, if you remember to turn the flashdrive off, it would protect you from unclean surges of electricity until you determine it safe to toggle it back on.

Of course, to get Lakka working again, you need to re-install the image on it if it was corrupted by a compromised supply of power.