libretro/Lakka-LibreELEC #229: No wireless connection

DraGiuS, 2017-09-30:
I use the 2.1 rc5 of lakka actually.

I need the ath10k driver for having wifi on my device. Just this.

Ntemis, 2017-09-30:
Thing is we don't know what your "device" is.

DraGiuS, 2017-09-30:
It's an Atheros QCA 9377 device.

ToKe79, 2017-12-09:
@DraGiuS The question is about the "Lakka device" - like generic PC (32/64 bit?) or a RPi / Odroid etc. with a USB WiFi.

DraGiuS, 2018-01-26:
Still no wireless on this device on a generic PC (Arch x86_64)

ToKe79, 2018-10-08:
@DraGiuS sorry for late response. Can you, please, try image from here: http://nightly.builds.lakka.tv/special_builds/ath10k/
I have added the ath10k driver/module (https://github.com/ToKe79/Lakka-LibreELEC/commit/6ca3efb42f5621b22c0eeef7c4bd8294ee8a5f47)

DraGiuS, 2018-10-13:
It seems to don't work.

ToKe79, 2018-10-22:
Please, provide output of: `cat /etc/release`, `lspci -nnk`, `lsusb`, `lsmod` and `dmesg`. Thank you!
See here how to connect to the command line: [Accessing Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/)

closed 2019-06-26 by ToKe79