libretro/Lakka-LibreELEC #230: All ScummVM stopped working RPi3/2.1RC4

masid101, 2017-10-02:
- Which version of Lakka are you using? 2.1 RC4
- What system hardware are you using? RPi3
- What did you do? Start a ScummVM game from the playlist
- What did you expect to happen? Game starts as in 2.0
- What happened instead? Fallback to retroarch

**Playlist entry ...**

/storage/roms/stick/scummvm/tentacle/MONSTER.SOU
Day of the Tentacle (CD_German)
DETECT
DETECT
BA6467A2|crc
ScummVM.lpl

**Here is the Log ...**

Lakka (community) Version: 2.1-rc4
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
RetroArch [INFO] :: This is RetroArch version 1.5.0 (Git 882fe09)
RetroArch [INFO] :: === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Jul 21 2017
RetroArch [INFO] :: Version: 1.5.0
RetroArch [INFO] :: Git: 882fe09
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [Config]: Loading default config.
RetroArch [INFO] :: [Config]: loading config from: (null).
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/.srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/.state".
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: [Audio]: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/tmp/shaders/slang-conversion-template.glsl"
RetroArch [INFO] :: Found shader "/tmp/shaders/stock.glsl"
RetroArch [INFO] :: [Video]: Video @ fullscreen
RetroArch [INFO] :: [Video]: Starting threaded video driver ...
RetroArch [INFO] :: [EGL] Falling back to eglGetDisplay
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: [GL]: Found GL context: videocore
RetroArch [INFO] :: [GL]: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: [GL]: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Linking GLSL program.
RetroArch [INFO] :: [GLSL]: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: [GLSL]: Found GLSL fragment shader.
[... 167 lines cut ...]
RetroArch [INFO] :: [udev] Adding device /dev/input/event0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev]: Plugged pad: PLAYSTATION(R)3 Controller (0:0) on port #0.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event1) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event1) supports 16 force feedback effects.
RetroArch [INFO] :: [Autodetect]: 186 profiles found.
RetroArch [INFO] :: [Autodetect]: selected configuration: /tmp/joypads/udev/Sony-PlayStation3-DualShock3-Controller-Bluetooth.cfg
RetroArch [INFO] :: [Joypad]: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: [Font]: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
scummvm: Unrecognized game target 'SOU'
Usage: scummvm [OPTIONS]... [GAME]

Try 'scummvm --help' for more options.
Lakka:~ # exit

RobLoach, 2017-10-02:
The ScummVM scanning/launching has changed, unfortunately. While the MONSTER.SOU detection was neat, it caused more problems than it solved. You'll have to [go through the Lakka ScummVM launching documentation](http://www.lakka.tv/doc/ScummVM/). Have a `tentacle.scummvm` text file with "tentacle" as the contents of it.

Having it be able to launch through MONSTER.SOU would be in https://github.com/libretro/scummvm/issues/78

masid101, 2017-10-02:
Ah, okay. I wasn't aware that the detection mode changed. Okay, will change to the new one. Thanks!

closed 2017-10-02 by masid101

masid101, 2017-10-02:
I just did what the new tutorial says. Then I ran detection. About 10 to 15 of 60 games were detected (All my games are certified by the Gruby database). It was the same meagre result as with the old version. With the old version I had to build most of the playlist by hand, now it seems the same!
After that I edited my old playlist to fit to the new system, every game works again!

What exactly changed now and what is the benefit of it?