libretro/Lakka-LibreELEC #231: [DOC] URL to issue tracker is obsolete

ToKe79, 2017-10-02:
In the online documentation (e.g. http://www.lakka.tv/doc/Home/) there is link to Issue tracker:
https://github.com/lakkatv/Lakka/issues

Correct URL:
https://github.com/libretro/Lakka-LibreELEC/issues

I have identified following places in the navigation menu (sidebar):
Contributing -> Fixing an issue
Support -> Bug tracker

I have checked several sub-pages and found no further links to issue tracker.

Note: when I open http://www.lakka.tv/doc/_Sidebar/ correct URL is on these items. Seems like the sidebar is not correctly pulled from _Sidebar.

Kivutar, 2017-10-03:
Hello, I fixed this a few days ago after your report. I think you need to ctrl+f5

ToKe79, 2017-10-03:
Hi, I tried ctrl+f5 / ctrl-r to refresh completely (not cached) before posting, but I still got the obsolete URL. Now I just tried different browser at work (IE11), which I did not use before to go to lakka.tv and same obsolete URL:
[image]

Kivutar, 2017-10-03:
Ok there was an issue in updating the website. This time it is fixed. Thanks for your help!

closed 2017-10-03 by Kivutar