libretro/Lakka-LibreELEC #232: Test 2.1-rc5 images for every platform

Kivutar, 2017-10-03:
Just check that the image boots, and can launch a rom.

- [ ] Generic.x86_64
- [ ] Generic.i386
- [ ] RPi.arm + noobs
- [ ] RPi2.arm + noobs
- [ ] a20.arm
- [ ] a10.arm
- [ ] Bananapi.arm
- [ ] imx6.cuboxi.arm
- [ ] imx6.udoo.arm
- [ ] OdroidC1.arm
- [ ] Odroid_C2.arm
- [ ] OdroidXU3.arm
- [ ] WeTek_Core.arm
- [ ] WeTek_Hub.arm
- [ ] WeTek_Play.arm
- [ ] WeTek_Play_2.arm
- [ ] Gamegirl.arm
- [ ] S8X2.arm
- [ ] S805.arm
- [ ] S905.arm
- [ ] H3.op2.arm
- [ ] H3.opipc.arm
- [ ] H3.opiplus.arm
- [ ] H3.opione.arm
- [ ] H3.opilite.arm
- [ ] H3.opiplus2e.arm
- [ ] H3.bpim2p.arm
- [ ] H3.bx2.arm
- [ ] TinkerBoard.arm

hutzero, 2017-10-06:
Is this for SD only or eMMC too?

Kivutar, 2017-10-06:
SD only

Kivutar, 2017-10-06:
XU3 looks broken, can somebody confirm?

jkms, 2017-10-06:
@Kivutar I can test XU3.

do you have an image? or should I just compile the current commit?

Kivutar, 2017-10-07:
@jkms Thanks, I just uploaded an image there http://static.kivutar.me/Lakka-OdroidXU3.arm-2.1-rc5.img.gz

jkms, 2017-10-07:
@Kivutar the image boots and I can load roms, however the samba server didn't work for me. had to use scp.

Kivutar, 2017-10-07:
Cool, thanks for your test.

hydraheim, 2017-10-09:
RPI Zero W boots and loads up ROMs just fine. There are issues with abnormal controller support for the DS4 though. Menu scrolling moves _way too fast_ and directional buttons only work some times. Reconnecting was needed several times just to get around the menu. Tested with a SNES usb controller and that worked fine. DS4 has worked fine in previous versions both on RPI Zero W and RPI3.

Kivutar, 2017-10-10:
Ok thanks, this DS4 problem affects all the modern kernels. We'll have to check why the kernel guys are keeping this regression.

Kivutar, 2017-10-11:
All images should be OK now.

closed 2017-10-11 by Kivutar

Kivutar, 2017-10-12:
Also with USB connection

Kivutar, 2017-10-12:
Mhhh, who checked the opipc line? Somebody here reports it broken: https://forums.libretro.com/t/lakka-2-1-rc5-released/12720/5

trulio, 2018-03-22:
Hi,

when i boot my MK808B-Plus on image Lakka-S805.MK808B-Plus.arm-2.1.img.gz,

strorage resize behave normally but once retroarch launch screen buffer is corrupted

I tried on a samsung TV but also on a full HD BENQ computer monitor.

I have the same behavior.

I also tried the MXQ 2.1 image with the same results 
(as kodi libreElec says MXQ and MK808 use the same image)

here are the files showing the issue:

Regards,

Trulio
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1839198/dmesg.txt)
[image]
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/1839199/log.txt)