libretro/Lakka-LibreELEC #234: DISTRO:libreelec

alorbach, 2017-10-05:
Is it possible to build with kodi support?
I am trying to build with mediaplayer="kodi" as well as DISTRO=LibreELEC for H3 devices (Orange PI). 

But compiling fails at some point with this error: 
`cedarx: no package.mk file found`

RobLoach, 2017-10-05:
Lakka's goal is to have just enough OS to run RetroArch. You may want to have a look at [escalade's LE8 Remix](https://forum.libreelec.tv/thread/302-le8-remix-generic-rpi-emulationstation-chrome-spotify-provide-logs-if-problems/): https://github.com/escalade/LibreELEC.tv as an alternative.

It boots into Kodi, and then you're able to launch RetroArch from there. You have to change a text file to switch it from launching emulation station, to just launch RetroArch.

Kivutar, 2017-10-06:
Or if you just need kodi on H3, there is certainly a libreelec port somewhere.
But we can't add Kodi support in lakka for now. Supporting RetroArch is already a huge amount of work.

closed 2017-10-06 by Kivutar

alorbach, 2017-10-06:
Thanks I am going to look into escalade's remix. Unfortunetally he doesn't have H3 project files like you ;). I am very interested in building for Orange PI h3 devices. But maybe it can be mixed. Would be great t have Kodi 17 running with RetroArch on top ;) on these devices.