libretro/Lakka-LibreELEC #237: Playlist symbols of MAME2003 still missing in 2.1 RC5

masid101, 2017-10-08:
Which version of Lakka are you using? 2.1 RC5 (http://static.kivutar.me/Lakka-RPi2.arm-2.1-rc5.img.gz)
    What system hardware are you using? RPi3
    What did you do? Installed my premade playlists
    What did you expect to happen? Having the playlist show the symbols
    What happened instead? Only showing black squares

masid101, 2017-10-12:
Seems the fix doesn't work ...

https://github.com/libretro/Lakka-LibreELEC/issues/201

Kivutar, 2017-10-13:
What icon theme are you using?

Kivutar, 2017-10-13:
Here you can see that the database is named like the icon: "MAME 2003"

https://github.com/libretro/libretro-database/blob/master/rdb/MAME%202003.rdb
https://github.com/libretro/retroarch-assets/blob/master/xmb/monochrome/png/MAME%202003.png

masid101, 2017-10-13:
Ah, someone put a blank between MAME an 2003. That's new!

closed 2017-10-13 by masid101