libretro/Lakka-LibreELEC #238: Keyboard not working while entering wifi password RPi3 2.1 RC5

masid101, 2017-10-08:
Which version of Lakka are you using? 2.1 RC5 (http://static.kivutar.me/Lakka-RPi2.arm-2.1-rc5.img.gz)
What system hardware are you using? RPi3
What did you do? Try to enter wifi password using keyboard
What did you expect to happen? Letters on the screen
What happened instead? Nothing

Using the virtual keyboard with the controller works!

Kivutar, 2017-10-10:
I confirm this bug

masid101, 2017-11-25:
Still not working with 2.1 stable release! As this is a very annoying and even milestoned bug, I don't really get it to release a stable without this being fixed!

GAZ082, 2017-11-25:
Just installed lakka in my cubox-i4 and can't believe this is happening. Please fix ASAP! If you need some extra info please let me know. Thanks!

Kivutar, 2017-12-10:
> Still not working with 2.1 stable release! As this is a very annoying and even milestoned bug, I don't really get it to release a stable without this being fixed!

I didn't want to block the release for a keyboard bug that affects only arm boards where the vast majority of our users are using joypad only.

This, combined to other factors like the fact that it was convenient for me to release 2.1 before my new job starts. It was either this, either 2.1 would still not be available.

@Ntemis pushed a potential fix for this. I'm going to test it as soon as I can. The 2.1 branch is going to get stability updates.

Kivutar, 2017-12-10:
@Ntemis the change doesn't fix the keyboard issue.

Ntemis, 2017-12-10:
Sad to hear @Kivutar will revert it. Thanks for the test

Kivutar, 2017-12-23:
I think it is due to this line https://github.com/libretro/RetroArch/blob/master/input/drivers/udev_input.c#L75

Kivutar, 2017-12-23:
Ok, this image have a fix, I'd like someone to try it please http://static.kivutar.me/unstable/Lakka-RPi2.arm-2.1-devel-20171222232758-r28013-g9f419ef.img.gz

masid101, 2017-12-23:
Sorry, but this fix have no effect here!

Kivutar, 2017-12-23:
Yeah :( it was not that...

masid101, 2017-12-23:
We already had a version where it was working. It was 2.0 or one of the RCs of 2.1. Can't remember the exact one. Just if this info is of any help ...!

masid101, 2017-12-25:
I did some research. It worked in 2.1 RC4.

ToKe79, 2018-01-25:
HI,
can someone please test this RPi2 build: http://nightly.builds.lakka.tv/RPi2-kbd-fix-WIP/
Thank you!

masid101, 2018-01-25:
Sorry, but no effect here!

ToKe79, 2018-01-26:
Thank you for testing.
If anybody can test this image - unfortunately I have no RPi:
http://nightly.builds.lakka.tv/RPi2-kbd-fix-WIP/02_second-try_RC4+retroarch_from_RC5+other_changes/
It is based on RC4, but updated RetroArch (as in RC5) + other minor changes in libretro packages (mostly updated version due to build failing).
Changes on top of RC4:
https://gist.github.com/ToKe79/56aaf0b99b7efa8195eb1414f4e01544

ToKe79, 2018-01-26:
And also an image with patched retroarch:
http://nightly.builds.lakka.tv/RPi2-kbd-fix-WIP/03_third-try_as_second_try+patched_RA/

Changes on top of RC4:
https://gist.github.com/ToKe79/6f1fe54fada1b234ff74500950fc17de

kivutar, 2018-01-26:
Thanks @ToKe79 
I tried your latest image, it still doesn't work.

ToKe79, 2018-01-26:
So back to square 1 - plain RC4 (with only updated melonds, assets, sameboy to fix building) to confirm that I am at the correct commit, that was working.
http://nightly.builds.lakka.tv/RPi2-kbd-fix-WIP/04_fourth-try_plain_RC4/
Thank you for testing.

ToKe79, 2018-01-30:
I made another test build for RPi2:
http://nightly.builds.lakka.tv/RPi2-kbd-fix-WIP/05_CurrentLakka_RetroArch_882fe09/
This time I only reverted retroarch package to the state as of RC4 and the rest of the image is current Lakka-2.1-dev branch (https://github.com/libretro/Lakka-LibreELEC/commit/07b402ca29ce2c32d5df2ef20803f3aacf595a49). If this works I will try to find specific commit that broke it (https://github.com/libretro/RetroArch/compare/882fe09...28397b3).
Compiled from https://github.com/ToKe79/Lakka-LibreELEC/commit/76198d729f2f189d1e4a22cc570cba5639cdbc94

masid101, 2018-01-30:
This one works here on RPi3.

ToKe79, 2018-01-30:
Thank you, I just confirmed same (works) also on S905. Please, stay tuned.

ToKe79, 2018-01-31:
Issue bisected to these commits: https://github.com/libretro/RetroArch/compare/5efb90d...1911aad

closed 2018-01-31 by ToKe79

mstroh76, 2018-02-02:
Lakka-RPi.arm-2.1-devel-r28180-g31ef195 and Pi Zero W:
works

Mutuduxf, 2018-03-03:
Lakka-RPi.arm-2.1.img and Pi Zero W:
still can't input the wifi password

ToKe79, 2018-03-03:
@Mutuduxf please use devel build from http://nightly.builds.lakka.tv

Shadow322, 2018-03-07:
@ToKe79  i dont see a raspi3 build at that link?

ToKe79, 2018-03-07:
Hi @Shadow322 
raspi3 = RPi2.arm
There is no "rpi3 optimized build", as there is no significant performance difference.

Shadow322, 2018-03-07:
oh ok ill stick with the better program called Retropie then thanks

starquake, 2018-03-07:
@Shadow322 If you think it fits your needs better then you should definitely switch. I hope you will be more friendly towards the retropie developers though.

PR #361 (2018-01-31): retroarch: Patch to fix keyboard typing
Tested on RPi3. The diff is straightforward.

Fixes #238
Fixes #239