libretro/Lakka-LibreELEC #242: CIFS mount not working in 2.1 RC5

hermanchunky, 2017-10-12:
In the latest RC CIFS mount don't seem to work anymore.

Following [this guide ](http://www.lakka.tv/doc/Serving-ROMs-from-a-NAS/) would work with the previous RC but not with RC5 anymore - not via sysctl script nor directly:

mount: mounting //192.168.1.125/MEDIA/roms on /storage/roms/nas/ failed: Operation not supported

Kivutar, 2017-10-12:
It's certainly because of the SAMBA 4 update.

Can you tell me if you're serving your roms from a Linux, Windows or Mac?

Kivutar, 2017-10-12:
Try this:

    mount -t cifs //192.168.1.125/MEDIA/roms /storage/roms/nas -o guest,uid=1000

hermanchunky, 2017-10-12:
Hi, I've supplied my credentials before in the sysctl script as well as when trying to mount it manually, resulting in the "Operation not supported" error.

If I try out with guest login (as you stated) the error is "Invalid argument". I'm not suprised because the share is not accessible by guest-login. I'm a bit suprised as the error message should be different with missing credentials, shouldn't it?

I've stored my roms on a Synology DS416.

hermanchunky, 2017-10-13:
I've changed the maximum SMB procotol version from SMB2 to SMB3. Now the mounting works fine again. Guess the issue has been resolved :-)

closed 2017-10-13 by hermanchunky

Kivutar, 2017-10-13:
So did you change something on the Lakka side to fix it? Or on the NAS side?

hermanchunky, 2017-10-13:
I changed it on the NAS side in the Synology DSM. Maybe LE only supports SMB3 upwards in the latest build!? The NAS' default setting was SMB2, so it didn't work...

Kivutar, 2017-10-13:
I was thinking that we were also compatible with SMB2.
@Ntemis what do you think about it? Is the kernel patch forcing smb3 or something?

Ntemis, 2017-10-14:
Its because samba 3 is compatible with 2.We force nothing but we demand > 1